micropython/micropython #1173: How to provide version in a CPython compatible way.

danicampora, 2015-03-31:
In https://github.com/micropython/micropython/commit/d460a307114285c675846bc4be2a57b59c06c11a
@dpgeorge commented:
> There was a discussion in #440 about how to provide the "version" in a CPython compatible way. The outcome of that discussion was to use sys.implementation (https://docs.python.org/3.3/library/sys.html#sys.implementation) but it was never implemented. I guess now is the time to do it. Eg sys.implementation would return: (name='micropython', version=(1.4.0)). We are free to add named fields to this tuple (as long as they begin with an underscore), so could add something like _board="pyboard" or _board="wipy" etc. Probably we need an issue to discuss this :)

I was not aware of that discussion, but I agree with the proposal. So, we can add `_board` to specify a board name, but how to include the version of the "port"?

danicampora, 2015-03-31:
`_board_version`? but that seems more like a hardware version.
`_port_version`?

dpgeorge, 2015-03-31:
Ok, we actually have a few existing things already implemented that I forgot about.  And there are some other options from the standard library that we can use for board/port version, git tag, build date, etc.

Currently implemented:
- `sys.platform`; current values used are "linux" and "pyboard" (need to define MICROPY_PY_SYS_PLATFORM for this)
- `sys.version`; currently gives "3.4.0"
- `sys.version_info`; currently gives `(3, 4, 0)`
- `os.name`; should have coarser granularity than `sys.platfarm`; currently gives "posix" on unix port

Not (yet) implemented:
- `sys.implementation`; should give something like `(name='micropython', version=(1, 4, 0))`
- `os.uname()`; gives details of the current operating system as a named tuple

I suggest using os.uname to provide detailed information about the hardware that uPy is running on.  This function returns a namedtuple with the following entries:
- sysname - operating system name; eg "pyboard", "wipy"
- nodename - name of machine on network (implementation-defined); could use hostname or unique ID of MCU
- release - operating system release; this would be the version of the port (eg "1.4.0" for stmhal, but "0.9.0" for wipy)
- version - operating system version; this can give git tag and build date (CPython on linux gives build date)
- machine - hardware identifier; this can give the board name and/or the chipset, eg "PYBv1.0 with STM32F405" (CPython on linux gives "x86_64" for me)

danicampora, 2015-04-01:
OK, I will define `MICROPY_PY_SYS_PLATFORM` for the WiPy.

> sys.implementation; should give something like (name='micropython', version=(1, 4, 0))
os.uname(); gives details of the current operating system as a named tuple

OK, lets implement those two.

> sysname - operating system name; eg "pyboard", "wipy"

Sounds good.

> nodename - name of machine on network (implementation-defined); could use hostname or unique ID of MCU

Yes, this can be the unique ID of the MCU or the MAC address in case of the WiPy.

> release - operating system release; this would be the version of the port (eg "1.4.0" for stmhal, but "0.9.0" for wipy)

OK

> version - operating system version; this can give git tag and build date (CPython on linux gives build date)

> machine - hardware identifier; this can give the board name and/or the chipset, eg "PYBv1.0 with STM32F405" (CPython on linux gives "x86_64" for me).

We can return something similar to what is printed on the welcome banner.

danicampora, 2015-04-01:
By the way, we will need a public `mp_obj_new_namedtuple` function.

dpgeorge, 2015-04-01:
> By the way, we will need a public mp_obj_new_namedtuple function.

They should be static const namedtuple's, not dynamically allocated ones.  So we need a way of making static const ones.

danicampora, 2015-04-01:
> They should be static const namedtuple's, not dynamically allocated ones. So we need a way of making static const ones.

OK, good point. But what about for methods like `WLAN.ifconfig()`, right now that one returns a dictionary, but might be better to return a named tuple instead. The same goes for the results of `WLAN.scan()` (list of namedtuples instead of list of tuples)

dpgeorge, 2015-04-01:
A namedtuple might be better than a dict.  A namedtuple instead of a tuple is a choice you have to make depending on user friendliness vs RAM usage.

For sys.implementation and os.uname(), they need to return namedtuples to be compliant with CPython stdlib.

danicampora, 2015-04-01:
Namedtuple is better than a tuple since both `WLAN.ifconfig()` and `WLAN.scan()`, return a lot of info which is difficult to read without field names. 
So, I propose to expose a `public mp_obj_new_namedtuple function` for those kind of things.

pfalcon, 2015-04-01:
> The same goes for the results of WLAN.scan() (list of namedtuples instead of list of tuples). Maybe we can discuss this somewhere else.

I raised that question re: esp826 port already: https://github.com/micropython/micropython/issues/876#issuecomment-71108167 , and that's apparently better place to discuss that.

pfalcon, 2015-04-01:
> sys.implementation and os.uname()

Do you guys consider difference between hardware and an OS running on it (which may be different)? Then quick thought is that sys.platform might be better for "hardware" an os.uname() for "OS". Didn't give thorough thinking to the topic of this ticket yet though.

dpgeorge, 2015-04-01:
> Do you guys consider difference between hardware and an OS running on it

Yes, definitely.  The hardware could be the PCB revision and MCU (eg PYBv1.0 with STM32F405), yet the OS for different hardware can be the same (eg pyboard).

os.uname() provides just the right API that we need: it has both OS name (sysname field) and hardware name (machine field).  See above.

dpgeorge, 2015-04-02:
To reduce RAM/ROM usage but still provide a namedtuple-like object (ie object with attribute accessors) we can implement a namespace object.  CPython has these and uses them for sys.implementation (among other things).  They are basically a dict but with attribute accessors instead of subscription, so you can do sys.implementation.name instead of sys.implementation['name'].

The field names could be shared among multiple instances.

Eg an instance of a namespace has an array of items, and a pointer to a (shared) array of qstrs with the same length as the array of items (so, like an ordered dict but the keys are in a separate array and shared).  It would cost a few bytes to implement, but would mean that things like sys.implemenation, os.name(), wlan.ifconfig, wlan.scan() (see #876) could be implemented as efficiently as possible.

danicampora, 2015-04-04:
OK sounds good to me.

dpgeorge, 2015-04-21:
Basics are now implemented in c3184aea63b6eaf0a0037f691927028ca4be84c2 (sys.implementation) and d8837cea6f395bb47863ef7db58ab3984215b7ff (os.uname() in stmahl).  Other ports should follow suit and implement os.uname().

closed 2015-04-21 by dpgeorge