micropython/micropython #1178: Strange floating point addition on Pyboard

peterhinch, 2015-04-10:
File rats.py:
```python
from math import pi, sin, cos
from array import array
length = 128
x = array('f', (cos(2*pi*x/length) for x in range(length)))
y = array('f', (sin(2*pi*x/length) for x in range(length)))
print(x[1], y[1]) # OK
print(x[1]*x[1], y[1]*y[1]) # OK
print(x[1]*x[1] + y[1]*y[1]) # ????
```
This produces the following output at the REPL
```
>>> import rats
0.9987954 0.04906768
0.9975924 0.002407637
0.0
>>> 
```
This behaviour only occurs when the expected result is precisely 1.0. It's not easy to replicate by simply typing in the values, presumably because of floating point conversion imprecision.

dpgeorge, 2015-04-10:
The problem is most likely with printf. The number is probably 0.999999999
and not exactly 1.0.

dpgeorge, 2015-04-11:
Here is a very concise test for this bug:
```python
print(0.9999999)
```

dpgeorge, 2015-04-11:
@dhylands @doogle do either of you mind looking into this?  The problem is with single precision floating point, related to this line here https://github.com/micropython/micropython/blob/master/py/formatfloat.c#L146.  To test, change unix build to use single precision instead of double and try the above print test.  Thanks!

dhylands, 2015-04-12:
I'll take a look t it. It fails in my original code as well.

dpgeorge, 2015-04-12:
Fixed.

closed 2015-04-12 by dpgeorge