micropython/micropython #1182: with + for + return = breakage

pfalcon, 2015-04-15:
Testcase:

~~~~
def test():
    with open("t1.py") as f:
        for l in f:
#        for i in range(10):
            return

test()
~~~~

Leads to: TypeError: 'TextIOWrapper' object is not callable . That's because where on stack WITH_CLEANUP expects `obj.__exit__`, there's actually for's iterator. Uncommenting range() leads to similar behavior - while there's no explicit iterator, loop control var (int) is on the stack.

dpgeorge, 2015-04-15:
When fixing this, the following nested version also needs to work:
```python
with open("t1.py") as f1:
    for l1 in f1:
        with open("t2.py") as f2:
            for l2 in f2:
                return
```
This will need to pop iterator f2, then execute the inner with, then pop iterator f1, then execute the outer with.

dxxb, 2015-10-13:
I have encountered this issue myself (see #1508) and after looking at vm.c, compile.c, and runtime.c I couldn't come up with a simple fix. When handling MP_BC_SETUP_WITH in vm.c `PUSH_EXC_BLOCK(1);` is executed and I thought MP_BC_RETURN_VALUE (see https://github.com/micropython/micropython/blob/master/py/vm.c#L1014) would result in the stack being returned to its state at the point in time when MP_BC_SETUP_WITH was executed. I clearly do not understand enough about how this should work.

What would be a good approach to fixing this?

dpgeorge, 2015-10-13:
> What would be a good approach to fixing this?

The problem is that the for loop retains its iterator on the stack and the unwinding does not know about this.  The best approach would be for the compiler to insert hints in the bytecode that there is an iterator that needs popping, or in fact add bytecodes to do the popping.  Eg in the above cases that fail the RETURN_VALUE opcode should become:
```
ROT_TWO # move iterator to top
POP_TOP # discard for iterator
RETURN_VALUE # return value (will lead to exception unwinding)
```
But this is just a sketch and may need further thought.

Note that break and continue already do something very similar, see py/emitbc.c:mp_emit_bc_unwind_jump.

pfalcon, 2015-10-13:
> The best approach would be for the compiler to insert hints in the bytecode

Well, I intuitively felt that dynamic unwinding won't work here and it would require handling on compiler side, that's why I didn't look into it beyond initial report and testcase. I'm glad you agree. And then it's what I hinted about in another ticket: either do something "overgood" like marry heap and non-heap floats in one implementation (both are good on their own, but together really hitting that "overgood" mark), or fix broken compiler. Tough choice ;-).

dxxb, 2015-10-14:
@dpgeorge, I do not think inserting `ROT+POP` would work in the nested case and neither would using the mechanism used by break/continue. Would wrapping for loops in exception blocks work instead? Upon return instructions emitted for the "finally clause" would pop the iterator.

dpgeorge, 2015-10-14:
No, it's even more difficult than I thought.  The FOR_ITER opcode pops the iterator at the end of the loop, so if you also popped it in a finally clause then it'd be popped twice if no return was executed.

It's probably going to need compiler and VM changes.

There are lots of test cases that need to be written, eg:
```python
with X():
    for a in [1,2]:
        for b in [3, 4]:
            with Y():
                for c in [5, 6]:
                    for d in [7, 8]:
                        return (a, b, c, d)
```
This needs to pop 2 iterators, then run the Y with, then pop 2 more, then run the X with, then return the value.

dxxb, 2015-10-14:
I didn't state it explicitly but I thought the finally clause would replace the current iterator pop in compile.c:compile_for_stmt(). That would work I think (yes, there would be compiler and VM changes involved). My question is: an implementation that wrapped every for loop in an exception block would be acceptable?

dpgeorge, 2015-10-14:
> I didn't state it explicitly but I thought the finally clause would replace the current iterator pop in compile.c:compile_for_stmt()

There is no iterator pop in the compiler.  It's done in the FOR_ITER opcode.

> My question is: an implementation that wrapped every for loop in an exception block would be acceptable?

No, too much overhead :)

dxxb, 2015-10-14:
> There is no iterator pop in the compiler. It's done in the FOR_ITER opcode.

my bad, I wrote that up from memory

> No, too much overhead :)

Ack. So I am out of ideas at the moment. If a solution is proposed I may try implementing it.

dpgeorge, 2015-10-15:
Fixed, see above.

Turned out that the VM had exactly the info needed: it knows the stack-pointer for all pushed exception handlers, including with-cleanup blocks.  So it knows the correct value of the stack-pointer that needs to be restored for the with-cleanup to execute, regardless of any for iterators that may exist.  So popping the for-iterators is very easily done by using the current sp and the known sp for the finally handler.

Having the compiler provide the information would have been a lot of work.

closed 2015-10-15 by dpgeorge

dxxb, 2015-10-15:
Thank you @dpgeorge

danicampora, 2015-10-15:
@dpgeorge you rock!

pfalcon, 2015-10-15:
@dpgeorge: Unbelievable!

dpgeorge, 2015-10-15:
Yeah, it's not a bad fix for such an insidious bug: effectively a 2 line fix, and costs only 16 bytes on x86-64 and Thumb2.