micropython/micropython #1187: Inline assembler data statement: alignment error causes crash

peterhinch, 2015-04-18:
The following causes a crash:
```
@micropython.asm_thumb
def bar():
    b(START)
    label(MYDATA)
    data(1, 111)
    label(START)
    mov(r0, 1)

print(bar())
```
This only occurs if the inline data occupies an odd number of bytes. I'd fix it but I can't figure out how the data statement works.

dpgeorge, 2015-04-18:
This is a feature, not a bug.  If you are going to put data within your code then you need to understand about instruction alignment.  Automatically aligning would be fighting against the programmer in this case.  There is the `align` instruction to do this for you:
```python
@micropython.asm_thumb
def bar():
    b(START)
    label(MYDATA)
    data(1, 111)
    align(2)
    label(START)
    mov(r0, 1)
print(bar())
```

closed 2015-04-18 by dpgeorge

peterhinch, 2015-04-18:
Fair point. Is `align` documented anywhere?

dpgeorge, 2015-04-18:
> Is align documented anywhere?

None of the inline assembler is (officially) documented.  Align takes 1 argument, being the number of bytes to align the next instruction/data to.