micropython/micropython #1188: ord(b'\x80') throws a TypeError

dhylands, 2015-04-19:
In unix micropython:
```
2012 >./micropython 
Micro Python v1.4.1-78-g404b68d on 2015-04-18; linux version
>>> ord(b'\x80')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: ord() expected a character, but string of length 0 found
```

In python3:
```
2013 >python3
Python 3.4.0 (default, Apr 11 2014, 13:05:11) 
[GCC 4.8.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> ord('\x80')
128
```

dhylands, 2015-04-19:
Originally reported on the forum: http://forum.micropython.org/viewtopic.php?f=6&t=651&p=3724#p3717

dpgeorge, 2015-04-19:
We had this issue before... `ord` is (implicitly) documented by CPython to work only for string (actually unicode) objects.  There's no mention of bytes/bytearray conversion.  Yet, as you found out, it does actually work for bytes (and bytearrays).  So we don't keep having this problem crop up I will fix it.

pfalcon, 2015-04-19:
@dpgeorge : Please make it #if CPYTHON_COMPAT

dpgeorge, 2015-04-19:
Actually, we do already support bytes to some extent, but it only works correctly for bytes chars < 0x80.  So the minimal fix is to just support bytes properly.  To have a CPYTHON_COMPAT option here would require a lot more code for the non-compatible version (since you need to then have an explicit check that it's a string object and issue an exception if it's not; currently this is done by mp_obj_str_get_data which raises an exception if the argument is not a str or bytes).

dpgeorge, 2015-04-19:
Fixed.  @pfalcon if you still think it needs a CPYTHON_COMPAT option then please say so.  Maybe we need to ask upstream about fixing the docs regarding this?

closed 2015-04-19 by dpgeorge

pfalcon, 2015-04-19:
Nope, looks good. And that's apparently a hidden py2 compat feature, probably not advertised on purpose ;-).