micropython/micropython #1189: Cleaning up wiki of unrelated pages

pfalcon, 2015-04-19:
Majority of pages in this wiki is catalog of some embedded boards which doesn't have any relation to MicroPython. These noise pages complicate access to other pages and leave impression of unmaintained project. If there's someone who maintains those pages, please come up and clarify the exact plan with them (like, if there's porting work for each of those boards, where patches will be submitted). Otherwise, I'm going to delete them in 1 week. Thanks.

Neon22, 2015-04-19:
I was maintaining them. Looks like they've served their purpose. delete away

pfalcon, 2015-04-19:
Thanks, I removed those for which ports didn't surface in over a year and which are not widely known. If you'll need that info for some other project, it's still available by cloning wiki's underlying git repo.

closed 2015-04-19 by pfalcon