micropython/micropython #1190: pyboard.py doesn't do encodings properly?

dhylands, 2015-04-19:
If you create a python file say c.py containing the following python code:
```python
print(chr(255))
```

And then try to use pyboard to run it on the pyboard, then you'll get the following error:
```
2079 >python3 pyboard.py c.py
Traceback (most recent call last):
  File "pyboard.py", line 247, in <module>
    main()
  File "pyboard.py", line 234, in main
    ret, ret_err = pyb.exec_raw(pyfile, timeout=None, data_consumer=lambda d:print(str(d, encoding='ascii'), end=''))
  File "pyboard.py", line 117, in exec_raw
    return self.follow(timeout, data_consumer)
  File "pyboard.py", line 86, in follow
    data = self.read_until(1, b'\x04', timeout=timeout, data_consumer=data_consumer)
  File "pyboard.py", line 49, in read_until
    data_consumer(data)
  File "pyboard.py", line 234, in <lambda>
    ret, ret_err = pyb.exec_raw(pyfile, timeout=None, data_consumer=lambda d:print(str(d, encoding='ascii'), end=''))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 0: ordinal not in range(128)
```

dhylands, 2015-04-19:
Originally from the forum post: http://forum.micropython.org/viewtopic.php?f=2&t=649

dpgeorge, 2015-04-19:
Thanks for the report.  Should now be fixed.

Note that there were 2 separate issues: receiving non-ASCII chars and sending non-ASCII chars through pyboard.py.  The exception given above had to do with receiving, which is fixed by properly handling 8-bit values in pyboard.py.  To fix sending (eg so you can write unicode chars in a script that is passed to the pyboard) required fixing pyexec.c, and so new firmware needs to be downloaded to fix that issue.

The following script works when run using pyboard.py:
```python
print(ord('α'))
print(chr(945))
```

closed 2015-04-19 by dpgeorge