micropython/micropython #1193: usb_mode docs seem to be incorrect/missing

dhylands, 2015-04-20:
This page: http://docs.micropython.org/en/latest/tutorial/usb_mouse.html references CDC+HID, but the code comments have 'VCP+HID':
https://github.com/micropython/micropython/blob/master/stmhal/usb.c#L182-L189

The actual code seems to support both.

I also couldn't find any documentation on pyb.usb_mode() in the docs website (the crux of this particular bug).

main.c generates a boot.py that uses CDC (and is missing the CDC only mode). If the intention is to switch to VCP then we should probably update that.

dpgeorge, 2015-04-21:
Yep, it needs to be documented.  This new USB code also needs thorough testing on Windows and Mac...

I plan to keep CDC and VCP as possible config names (aliasing to the same thing) for backwards compatibility reasons.  But I agree that the fresh boot.py should be updated.

dpgeorge, 2016-08-29:
Fixed by above.

closed 2016-08-29 by dpgeorge