micropython/micropython #1196: Official mirrors on other hosting services

pfalcon, 2015-04-24:
Open-source software hosting space contracts, and github becomes way to crucial point for too many projects - both technically and organizationally. And that's of course risky. Can we please get an official git mirror on some other hosting sites, or at lease one? I actually had my own mirror on gitorious, which is about to be closed, and recently set up one on bitbucket. With 2 services dying at the same time (gitorious & googlecode), I actually feel uneasy still, and next step would be to set project on sourceforge, but of course I don't want register it myself, and would prefer that we have it organized.

dhylands, 2015-04-25:
I'm not sure why you would feel uneasy. Everyone of us that has micropython checked out essentially has a backup.

So if github went down, then you'd stop getting updates until either github came back up, or another publically available repository was available.

I have my micropython repository sitting in a DropBox account here:
https://www.dropbox.com/sh/9jndge1aji3vtpf/AABqXF-TbQjAvWt6TLvHjEJWa?dl=0
so you can always use that as a mirror. It's usually no more than a week old.

Anybody that's really concerned can always setup their own local mirror.

At work, our builds reference several hundred repositories on github, and we have hundreds of developers that use those. So we created our own local mirrors of those hundreds of repositories so that our automated builds keep chugging along even during those brief periods that github goes down or otherwise becomes inaccessible.

pfalcon, 2015-04-25:
> until either github came back up, or another publically available repository was available.

There's no need to wait "until", the talk is about providing mirror proactively.

> Anybody that's really concerned can always setup their own local mirror.

Just to clarify, the talk is about "global" mirrors.

As I mentioned, I have mirrors on 2 services, and looking for more (once again, there're 2 services shutting down right now, 3 - 2 = 1, aka no backup, so having 3 copies is not enough in our digital age of longevity problems). And next suitable place has 1-level namespace, so if I register "micropython", nobody else can, and I of course step aside to let @dpgeorge do it. If he's not interested/too busy for that, I can do it myself and add him later.

dpgeorge, 2015-04-26:
So the main reason for mirroring is so we have a backup for the "github-like collaboration & workflow", and not for actually backing up the code, right?

pfalcon, 2015-04-26:
Nope, only for the code - it would be nice to explicitly mark those projects as mirror-only and refer to github. Suggestion to use well-known hosting sites is to improve the reach (and secondarily, of course, to allow mirror discovery and serve as collaboration backup).

If this all sounds too farfetched and paranoid, just thing about it - soon a lot of code will be gone, completely, taken down my googlecode (they seem to make an arrangement that gitorious will become r/o, but who knows how long that will exist either).

If that still doesn't ring a bell, here's another anecdote (I have more than one actually): when I first heard about Espruino (long before its Kickstarter, when it was closed source), I made an investigation: a) what the guy could have used if hadn't develop a JS interpreter from scratch; b) what could be used for similar purpose, if Espruino will never have gone open-source. I found mentioning of "SEE" interpreter, but the source seem to have been gone from all places! By pure chance I found it here on github, and mirrored myself: https://github.com/pfalcon-mirrors/SEE .

dpgeorge, 2015-04-26:
Ok, so it's for global longevity (not reliant on any of us to provide the code to someone else) and partially for further reach into the community.

The only objection I have is that it's not the best fit to have sourcforge as a r/o mirror (that's not sourceforges intented activity).  How does one push changes automatically?  What if people start "commenting" at sourceforge?  I have no time to look there and here.

pfalcon, 2015-04-26:
> Ok, so it's for global longevity (not reliant on any of us to provide the code to someone else) and partially for further reach into the community.

Yes. Actually, I wanted to bring that issue up for "further reach" long time ago, up to forgetting about that. Remembered the issue now because of services going down.

> it's not the best fit to have sourcforge as a r/o mirror (that's not sourceforges intented activity

SourceForge is the oldest open-source hosting site, still active. When I worked there, informal rendition of hosting ToS was "please don't host yet another gigabyte binary linux distro with us". I don't think they would ban us for brining us more content to them. I understand that hosting extra copies on many sites is not the best usage of world's resources, but they host a lot of once-off stuff for years. Also, hosting services play games with source code (like shutting it down), let source code play games with them.

> How does one push changes automatically?

Add few git remotes, makes a script which pushes to them, run it from time to time. (There's always cron for pedants.)

>  What if people start "commenting" at sourceforge? I have no time to look there and here.

That's why nice to say it's just a source code repo, please submit bugs/patches on github. Disabling corresponding facilities help. Then, anything which gets thru can either be ignored, or someone else can review it few times a year and reply "sorry, please use ..."

stinos, 2015-04-27:
> Add few git remotes, makes a script which pushes to them, run it from time to time. (There's always cron for pedants.)

I thought github has the possibility for post-receive hooks just like client side git? That would be by far the easiest way to do this

dpgeorge, 2015-05-03:
Cloned at http://sourceforge.net/projects/micropython

SF now have an auto github sync thing which apparently syncs each release, and I enabled it.  Also I pointed the "support" link to the issue page at github.

closed 2015-05-03 by dpgeorge

pfalcon, 2015-06-13:
@dpgeorge added me to sf.net project, so I will push to it from time to time (in a handmade way).