micropython/micropython #1204: If an error occurs while interrupts are disabled, no report shows on the REPL

dhylands, 2015-04-28:
If you call pyb.disable_irq() and then encounter a syntax error, you don't get to see on the USB repl.

If you hook a UART and call pyb.repl_uart then you will see the error on the uart REPL, and if you type pyb.enable_irq() on the UART repl, then the error output will show up on the USB repl.

If we get back to the repl and interrupts are disabled, perhaps we should re-enable them? Just so that the error can be reported?

dpgeorge, 2015-04-28:
I tend to agree.  It's a usability issue more than anything.  Sure you can tell people to use try/finally or a context manager for disabling irqs (see #1016) but that doesn't solve the real problem, namely that people make mistakes and in this case a mistake with the irqs disabled is pretty hard to recover from.

The logic would be: if USB enabled and VCP mode selected and at friendly REPL prompt and irqs disabled, then enable irqs.  And I'd also like to print a message (eg "enabling irqs") to make it clear what happened.

This would also fix the "problem" that if you type `pyb.disable_irq()` at the REPL then it hangs...

dpgeorge, 2015-04-29:
Fixed.

closed 2015-04-29 by dpgeorge