micropython/micropython #1207: Add Lin-Bus support

zabeloliver, 2015-04-29:
Hi, 

beside the already implemented CAN-Bus it would be great to have the LIN-Bus functionality as well. With this you could open the uPy to more applications.

dpgeorge, 2015-04-29:
I don't have any familiarity with LIN busses, but from reading wikipedia it seems you can implement the protocol using a UART.  If so, the driver could probably be written in pure Python.

danicampora, 2015-04-29:
> I don't have any familiarity with LIN busses

I do, I worked in automotive for a few years. The tricky part is going to be measuring the pulses of the SYNC byte to adapt dynamically to the baud rate (slaves must do this). A dedicated LIN peripheral takes care of that part. Should be doable in pure Python on the pyboard.

zabeloliver, 2015-04-29:
So if i'm right, i could access all what i need in pure python and with the right HW-interface it should be possible to implement the LIN functionality?

danicampora, 2015-04-29:
Yes, you need a LIN transceiver. One like this:
http://www.atmel.com/Images/Atmel-4986-LIN-Networking-ATA6622C-ATA6624C-ATA6626C_Datasheet.pdf

pfalcon, 2015-05-26:
Thanks, we're all for supporting all existing buses in MicroPython, not just LIN-Bus. We however try to keep number of tickets in the bugtracker manageable, so feel free to discuss this feature on the forum http://forum.micropython.org or submit pull requests.

closed 2015-05-26 by pfalcon

habkorn, 2017-05-07:
Hi, do you have an minimal example showing how to use LIN bus using MicroPython?