micropython/micropython #1208: struct unpack() error handling

peterhinch, 2015-04-30:
The following code is accepted by MicroPython but throws an error in Cpython ("unpack requires a bytes object of length 2")
```python
from struct import unpack
z = bytes([1])
unpack('<H',z)
```

dpgeorge, 2015-04-30:
Well, this is a feature of it being "micro": it assumes you gave the correct sized buffer.  It would cost bytes to check and raise the appropriate error.

peterhinch, 2015-05-02:
Fair enough.

pfalcon, 2015-05-02:
Well, we want to know (and catalog) issues people faced, but actually fixing that will likely required django criteria per https://github.com/micropython/micropython/wiki/ContributorGuidelines .

pfalcon, 2015-05-04:
See also #1220.

dpgeorge, 2017-09-08:
This is now fixed, unpack will raise an exception if the buffer is too small (but not if it's too long).

closed 2017-09-08 by dpgeorge