micropython/micropython #1216: Object methods don't check "self" argument type (consistently)

pfalcon, 2015-05-03:
Split from https://github.com/micropython/micropython/issues/1203#issuecomment-98546768 :

> this general problem (whose archetypal example of a crash is list.append(1, 2)).

pfalcon, 2015-05-03:
> BTW, it may not be that hard to provide a generic check for the "self" argument to guard against list.append(1, 2). For load_method opcodes no check is needed because you are always guaranteed that "self" is the correct type. It's only when load_attr is called to get a non-static and non-class method (ie an instance method) that checking needs to take place on the call. These cases are rare and we could wrap the returned function in a checker. (I think this is all related to your question on py-dev about magic static methods: https://mail.python.org/pipermail/python-dev/2015-March/138950.html)

pfalcon, 2015-05-03:
My comment on why other language implementations don't have this problem is that they usually have some predefined way to check/parse methods args - either as part of method definition structure, or as convention to be used as a first thing in method code. This usually handled all args, including self/this automagically. Of course, this adds overhead, that's why we didn't resolve this issue before, and would like to resolve it efficiently.

The fact that we don't need it for "lexical" calls sounds good, but we still need it for general case. The simplest, but cumbersome way is to add explicit checks to each method, but then we have overhead for all cases. Otherwise, we need a way to tell if given function object is a method (i.e. requiring self arg), and then what type that method of, so we can check "self" compatibility.

To mark a method we either need a bit in method def structure, or some other scheme, like group methods together (vs non-instance-methods). The latter will then require something like pointer to a split point.

Telling a type would be even more complicated. I thought that we for example can put a backpointer to a type after the last element in method def table. Then, given a method pointer (which is in a table), we can walk the table till this last element and figure its type in such way.

dpgeorge, 2015-06-19:
See #1340 for a proposed fix for the general case.

dpgeorge, 2015-06-22:
Fix merged.

closed 2015-06-22 by dpgeorge