micropython/micropython #1217: USB HID Receive on stm32F4Discovery

452, 2015-05-04:
How to receive HID data in my stm32F4Discovery board with Micro Python?
please can you provide some Docs link or example with receiving and callback (onReceive)?

```python
pyb.usb_mode("VCP+HID", pid=0x1234, hid=(0, 0, report_len, 50, hid_desc))
```
when I double pressed tab button I have only one function:
```python
pyb.USB_HID.send
```
receiving data from host device like PC or tablet not implemented yet on stm32F4Discovery board?

```python
#work good
i=0
while True:
	pyb.USB_HID.send(0,"{} Yet another long text".format(i))
	pyb.delay(100)
	i+=1
```

I develop https://play.google.com/store/apps/details?id=com.appspot.usbhidterminal

about board and Micro Python version info, how to get it(with sudo screen /dev/ttyACM0 I have no info, only blank screen, after hit enter, I see:
```python
AT

Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 
>>> 9+8
17
>>> 
PYB: sync filesystems #PRESSED CRTL+D
PYB: soft reboot
Traceback (most recent call last):
  File "main.py", line 9, in <module>
KeyboardInterrupt: #PRESSED CRTL+C
Micro Python v1.4.2-55-g2378be4 on 2015-05-03; F4DISC with STM32F407
Type "help()" for more information.
>>> 
```
```python
ihor@ihor-M50Vc:~$ sudo picocom /dev/ttyACM0
picocom v1.7

port is        : /dev/ttyACM0
flowcontrol    : none
baudrate is    : 9600
parity is      : none
databits are   : 8
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Terminal ready
~x~ # PRESSED Enter
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 
```
, and then command prompt >>>)?
please add pyb.version(), and help() support with:
(about version, number of version integer value, full info - text)
```
Micro Python v1.4.2-55-g2378be4 on 2015-05-03; F4DISC with STM32F407
Type "help()" for more information.
```

and Thank you, Micro Python is awesome, good one =), I did not like Python, but now my view is changing =)

dhylands, 2015-05-04:
I see an AT command at the very beginning of your log. I know that minicom, by default, assumes that its talking with a modem, and sends a bunch of initialization strings.

I normally use picocom which doesn't send any of those types of strings. I see you used picocom in your second invocation, but it doesn't show what you used in the first invocation.

452, 2015-05-05:
@dhylands in first invocation I used:
```sh
sudo screen /dev/ttyACM0
```

dhylands, 2015-05-05:
I haven't used screen much at all. Perhaps you have some commands in your .screenrc which are sending the AT commands. I think that's what's causing the SyntaxError when you connect.

452, 2015-05-05:
I have no any settings for screen... I think..
```sh
ihor@ihor-M50Vc:~$ sudo tree /etc/screenrc
[sudo] password for ihor: 
/etc/screenrc [error opening dir]

0 directories, 0 files
ihor@ihor-M50Vc:~$ ls /etc/etc*
ls: cannot access /etc/etc*: No such file or directory
ihor@ihor-M50Vc:~$ ls /etc/scr*
/etc/screenrc
ihor@ihor-M50Vc:~$ ls /home/ihor/scr*
ls: cannot access /home/ihor/scr*: No such file or directory
ihor@ihor-M50Vc:~$ ls /home/ihor/.scr*
ls: cannot access /home/ihor/.scr*: No such file or directory
```

452, 2015-05-05:
```sh
ihor@ihor-M50Vc:/$ cat /etc/screenrc
# $Id: screenrc,v 1.15 2003/10/08 11:39:03 zal Exp $
#
# /etc/screenrc
#
#   This is the system wide screenrc.
#
#   You can use this file to change the default behavior of screen system wide
#   or copy it to ~/.screenrc and use it as a starting point for your own
#   settings.
#
#   Commands in this file are used to set options, bind screen functions to
#   keys, redefine terminal capabilities, and to automatically establish one or
#   more windows at the beginning of your screen session.
#
#   This is not a comprehensive list of options, look at the screen manual for
#   details on everything that you can put in this file.
#

# ------------------------------------------------------------------------------
# SCREEN SETTINGS
# ------------------------------------------------------------------------------

#startup_message off
#nethack on

#defflow on # will force screen to process ^S/^Q
deflogin on
#autodetach off

# turn visual bell on
vbell on
vbell_msg "   Wuff  ----  Wuff!!  "

# define a bigger scrollback, default is 100 lines
defscrollback 1024

# ------------------------------------------------------------------------------
# SCREEN KEYBINDINGS
# ------------------------------------------------------------------------------

# Remove some stupid / dangerous key bindings
bind ^k
#bind L
bind ^\
# Make them better
bind \\ quit
bind K kill
bind I login on
bind O login off
bind } history

# An example of a "screen scraper" which will launch urlview on the current
# screen window
#
#bind ^B eval "hardcopy_append off" "hardcopy -h $HOME/.screen-urlview" "screen urlview $HOME/.screen-urlview"

# ------------------------------------------------------------------------------
# TERMINAL SETTINGS
# ------------------------------------------------------------------------------

# The vt100 description does not mention "dl". *sigh*
termcapinfo vt100 dl=5\E[M

# turn sending of screen messages to hardstatus off
hardstatus off
# Set the hardstatus prop on gui terms to set the titlebar/icon title
termcapinfo xterm*|rxvt*|kterm*|Eterm* hs:ts=\E]0;:fs=\007:ds=\E]0;\007
# use this for the hard status string
hardstatus string "%h%? users: %u%?"

# An alternative hardstatus to display a bar at the bottom listing the
# windownames and highlighting the current windowname in blue. (This is only
# enabled if there is no hardstatus setting for your terminal)
#
#hardstatus lastline "%-Lw%{= BW}%50>%n%f* %t%{-}%+Lw%<"

# set these terminals up to be 'optimal' instead of vt100
termcapinfo xterm*|linux*|rxvt*|Eterm* OP

# Change the xterm initialization string from is2=\E[!p\E[?3;4l\E[4l\E>
# (This fixes the "Aborted because of window size change" konsole symptoms found
#  in bug #134198)
termcapinfo xterm 'is=\E[r\E[m\E[2J\E[H\E[?7h\E[?1;4;6l'

# To get screen to add lines to xterm's scrollback buffer, uncomment the
# following termcapinfo line which tells xterm to use the normal screen buffer
# (which has scrollback), not the alternate screen buffer.
#
#termcapinfo xterm|xterms|xs|rxvt ti@:te@

# Enable non-blocking mode to better cope with flaky ssh connections.
defnonblock 5

# ------------------------------------------------------------------------------
# STARTUP SCREENS
# ------------------------------------------------------------------------------

# Example of automatically running some programs in windows on screen startup.
#
#   The following will open top in the first window, an ssh session to monkey
#   in the next window, and then open mutt and tail in windows 8 and 9
#   respectively.
#
# screen top
# screen -t monkey ssh monkey
# screen -t mail 8 mutt
# screen -t daemon 9 tail -f /var/log/daemon.log
```

prusnak, 2016-09-22:
Implemented in https://github.com/micropython/micropython/pull/2448

closed 2016-10-04 by dpgeorge