micropython/micropython #1218: rfduino support request

452, 2015-05-04:
Please add RFD22121 and RFD22102 board support to MicroPy =)

http://www.rfduino.com/product/rfd22102-rfduino-dip/index.html#tab-description
http://www.mouser.com/ds/2/470/rfd22301.data.sheet.11.24.13_11.38pm-272240.pdf

```
MCU - 16MHZ ARM Cortex Mo
Flash -128kb
RAM- 8kb
Bluetooth 4.0 low energy radio
ARM Cortex M0 Microprocessor
Programmed via Arduino IDE
MCU includes ADC, I²C, SPI, UART and GPIO
```

pfalcon, 2015-05-04:
Sorry, such requests don't make sense - we want to support **all** boards in MicroPython even without them. But we're only humans, and can realistically support only few. If you need that board, please submit patches for it, and join the project as its maintainer.

closed 2015-05-04 by pfalcon

dpgeorge, 2015-05-04:
The forum (http://forum.micropython.org) is a better place to discuss these things.  You may find people there willing to help (especially if you send them free hardware!).