micropython/micropython #1220: Need to free "struct" module name for a compliant implementation

pfalcon, 2015-05-04:
We aren't going to add much to builtin "struct" module, but it lacks a lot to run existing python code. Thus, we need to rename it to "ustruct" to give way to Python-level implementation of "struct".

dpgeorge, 2015-05-04:
> it lacks a lot to run existing python code

Examples?

pack_into and unpack_from should be part of the core, since they allow to use it without allocating RAM.

pfalcon, 2015-05-04:
~~~~
$ micropython -X heapsize=8M scratchabit.py micropython.def 
Warning: /home/pfalcon/.micropython/lib/email/__init__.py is imported as namespace package
Warning: /mnt/hdd/projects/ScratchABit/pyedit/__init__.py is imported as namespace package
Warning: plugins/cpu/pymsasid3/__init__.py is imported as namespace package
Loading CPU plugin x86_32_pymsasid
Warning: plugins/loader/pyelftools/__init__.py is imported as namespace package
Traceback (most recent call last):
  File "scratchabit.py", line 460, in <module>
  File "scratchabit.py", line 421, in parse_disasm_def
  File "scratchabit.py", line 397, in parse_disasm_def
  File "plugins/loader/elf.py", line 1, in <module>
  File "plugins/loader/pyelftools/elftools/elf/elffile.py", line 11, in <module>
  File "plugins/loader/pyelftools/elftools/common/utils.py", line 12, in <module>
  File "plugins/loader/pyelftools/elftools/construct/__init__.py", line 33, in <module>
  File "plugins/loader/pyelftools/elftools/construct/core.py", line 1, in <module>
ImportError: cannot import name Struct
~~~~

As can be seen, happens when executing https://github.com/eliben/pyelftools

dpgeorge, 2015-05-04:
Ok, you need the Struct class wrapper.  I agree that should go as a pure Python implementation.

pfalcon, 2015-05-04:
Posted #1221

pfalcon, 2015-05-06:
Fixed.

closed 2015-05-06 by pfalcon