micropython/micropython #1222: NotImplemented not implemented

pfalcon, 2015-05-04:
https://docs.python.org/3/library/constants.html?highlight=notimplemented#NotImplemented

>    Special value which should be returned by the binary special methods (e.g. __eq__(), __lt__(), __add__(), __rsub__(), etc.) to indicate that the operation is not implemented with respect to the other type; may be returned by the in-place binary special methods (e.g. __imul__(), __iand__(), etc.) for the same purpose. Its truth value is true.

Given that we have Ellipsis, I guess we won't get around implementing this. (Alias it to MP_OBJ_SENTINEL ?)

pfalcon, 2015-05-04:
Proposed patch: #1223

dpgeorge, 2015-05-05:
Ellipsis is there because it's also a token (...).

I assume you came across code that really needs to use NotImplemented.  I'm happy to just copy the Ellipsis implementation for this.

pfalcon, 2015-05-05:
> I assume you came across code that really needs to use NotImplemented.

Actually, not. I just try to consider whether a uPy-compatible fix is mergeable upstream or not. I doubt that someone would agree to replace NotImplemented with Ellipsis name, even though the semantics is the same.

Also, reading how comparisons on user types are implemented in Python3, we *need* to support NotImplemented.

http://jcalderone.livejournal.com/32837.html , etc.

pfalcon, 2015-05-05:
Implemented, #ifdef'able.

closed 2015-05-05 by pfalcon