micropython/micropython #1226: Passing kwargs to a method which references super() is broken

pfalcon, 2015-05-04:
Tescase:

~~~~
class A:

    def __init__(self, foo):
        #super().__init__()
        super()
        pass

A(foo=32)
~~~~

Instrumented to print a culprit "exisiting" value, the output is:

~~~~
<A object at 9c66df40>existing val: 
Traceback (most recent call last):
  File "a.py", line 9, in <module>
TypeError: function got multiple values for argument 'foo'
~~~~

I.e. where argument setup expects "foo" to be located, there's actually value for "self". Commenting out "super()" makes issues go away.

pfalcon, 2015-05-05:
@dpgeorge , as this apparently affected by super() compiling/implementation, and I don't know much about it, could you queue up looking into this? I hit this in my own code, so can work it around, but it's pretty bleak, and in general case, it's not possible.

dpgeorge, 2015-05-05:
ok

dpgeorge, 2015-05-05:
It's not about super, it's about kwargs to a closed over function:
```
def f():
    x = 1
    def g(z):
        print(x, z)
    return g
f()(z=42)
```

dpgeorge, 2015-05-06:
Fixed.

closed 2015-05-06 by dpgeorge

pfalcon, 2015-05-06:
Great, thanks!