micropython/micropython #1236: [wiki] add sidebar to the wiki, organize

cloudformdesign, 2015-05-06:
The wiki really needs a sidebar to help keep everything organized. I volunteer to do this, if you give me the necessary permissions. I've worked on several github wikis for my job, and am very familiar with how to organize them.

pfalcon, 2015-05-07:
Please submit patches for review.

cloudformdesign, 2015-05-07:
here is my initial patch:

```
From 6270b139b571dfdfa460e4105a1937ad6e971789 Mon Sep 17 00:00:00 2001
From: Garrett Berg <garrett@cloudformdesign.com>
Date: Thu, 7 May 2015 08:10:48 -0600
Subject: [PATCH] add sidebar

---
 _Sidebar.md | 23 +++++++++++++++++++++++
 1 file changed, 23 insertions(+)
 create mode 100644 _Sidebar.md

diff --git a/_Sidebar.md b/_Sidebar.md
new file mode 100644
index 0000000..f7fc58c
--- /dev/null
+++ b/_Sidebar.md
@@ -0,0 +1,23 @@
+# Getting Started
+- [[Installation|Getting Started]]
+- [[Learn Micropython|Learn Micropython]]
+- [[Differences to CPython|Differences]]
+- ~~[[Examples|Examples]]  (needs cleanup and better beginner examples)~~
+- [micropython-lib](https://github.com/micropython/micropython-lib)
+
+## Python Boards and MCUs
+- [The pyb module](http://docs.micropython.org/en/latest/library/pyb.html)
+- [[Boards Summary|Boards Summary]]
+- [[Hardware Details|Hardware Details]] (need to compile pwm timers, servos, etc.)
+
+# General info
+- [[Performance|Performance]]
+- [[Comparison to ELua|MicroPythonVsELua]]
+
+# Development
+- [[Contributor Guidelines|ContributorGuidelines]]
+- [[Development Workflow|DevelWorkflow]]
+- [[Internals|Internals]]
+- [[ARM Toolchain|Getting Started STM]]
+
+
-- 
1.9.1
```

The patch includes a few dead links on purpose -- they need to be added.

Doing things this way is very cumbersome, it is much easier for me to edit wiki's when I have the ability to control the sidebar. My normal process is to:

- organize what is available by topic and when a new user would be interested in it
- add dead links for what needs to be added
    - in the github wiki, when you click on a dead link it automatically creates the page and you can start editing. Also, the dead link comes up as RED, so you know it's dead and a "wiki TODO"
- edit pages so that the whole wiki can be cohesive.

This is much more difficult when I have to submit single patches -- I can't really do the whole workflow.

dpgeorge, 2015-05-07:
I think anyone can edit the github wiki.

cloudformdesign, 2015-05-07:
That's what I've been saying, I can't edit the sidebar!

dpgeorge, 2015-05-07:
Neither can I it seems.  Maybe we need to add a custom sidebar?

cloudformdesign, 2015-05-07:
on my wiki it is just a button below the default sidebar -- that button isn't there for me

[image]

dpgeorge, 2015-05-07:
I added a custom sidebar, but now it seems to automatically collapse the original one.

cloudformdesign, 2015-05-07:
that's because you have to edit it manually. Look at it now!

dpgeorge, 2015-05-07:
Ok, that's definitely better!

cloudformdesign, 2015-05-07:
alright, I've got everything I need. Closing this issue!

closed 2015-05-07 by cloudformdesign

pfalcon, 2015-05-07:
Please remove "Comparison to ELua" from sidebar - that page is essentially my personal notes and not ready to go public (needs style to be adjusted to be more neutral).

pfalcon, 2015-05-07:
> Doing things this way is very cumbersome

github wiki is based on git, so you can just make changes any way you like and give link which shows diffs. Here's example: https://github.com/pfalcon/esp8266-wiki/wiki/Memory-Map/_compare/3debc5aa8c69a525101490cd446b2aed98e750ec...1f28d4e19927bbaa9ff60f5dab87d2d950cf1b7d . Anyway, I'm glad you figured a way to do it.

cloudformdesign, 2015-05-07:
Feel free to remove it yourself! I am not the king of the wiki :)

Just click the edit button next to the sidebar on any wiki page and delete that line.

pfalcon, 2015-05-07:
> Feel free to remove it yourself! I am not the king of the wiki :)

And yet you ignored suggestion to pass such massive change thru review first. And now you suggest someone else should clean it up.