micropython/micropython #1237: esp8266: list()/accept() broken?

roger-, 2015-05-07:
I'm getting `OSError: not listening` with the following server test script:

```python
import esp

esp.connect('xxx', 'xxx')  
 
s = esp.socket()
  
def on_connect(sock):
    print(s.accept())
    
    conn, addr = s.accept()
    print('Connected with ' + addr[0] + ':' + str(addr[1]))
     
s.bind(('', 80))
s.onconnect(on_connect)
s.listen(5)
```

Seems like something is broken?

atalax, 2015-05-07:
> ```python
> def on_connect(sock):
>    print(s.accept())
> ```

Actually, the child socket gets passed as the argument, so you should do
```python
def on_connect(sock):
    print(sock) 
```

Simple HTTP server:
```
import esp
import sys

def onnewclient(s):
    print("Client connected!")
    s.onrecv(onclientdata)
    s.onsent(onclientsent)

def onclientdata(s, d):
    dp = d.split()
    path = dp[1].format("ascii")
    if path == "/":
        s.send("HTTP/1.0 200 OK\r\n"
               "Server: Micropython for ESP8266\r\n"
               "Content-Type: text/html\r\n"
               "Connection: close\r\n"
               "\r\n"
               "<h1>ESP8266</h1>\r\n"
               "Micropython version " + str(sys.implementation[1]))
    else:
        s.send("HTTP/1.0 404 OK\r\n"
               "Server: Micropython for ESP8266\r\n"
               "Content-Type: text/html\r\n"
               "Connection: close\r\n"
               "\r\n"
               "<h1>:(</h1>")

def onclientsent(s):
    print("Closing...")
    s.close()

server = esp.socket()

server.onconnect(onnewclient)
server.bind(("", 80))
server.listen(5)
```

roger-, 2015-05-07:
Cool, so we don't need to use accept/recv? Are those even useful to have then?

deshipu, 2015-07-03:
Is this still an open issue, or should we close it?

pfalcon, 2015-07-03:
As there's no standard test procedure for this port, it's hard to tell if it's fixed or not, and if/when it may be broken again.

deshipu, 2015-07-03:
Maybe I'm reading this wrong, but it seems to me, that the issue itself is invalid. The attached code should never have worked, and the the code given in the comment is the valid way to do that instead, which, I can confirm, works. So the whole issue was a misunderstanding on the user side?

roger-, 2015-07-03:
The issue was that the API was a bit confusing.

I'm still not sure what `accept()`/`recv()` are used for though.

dhylands, 2015-07-03:
In the normal socket stack, you do a listen, and when somebody eventually connects, then the server side will get some information about the connector. If that information (typically IP address) is deemed to be acceptable then the server side would do an accept. The socket returned by listen is what's called a listening socket that's not bound to any particular remote, and when you call accept, you get a new socket which is actually bound to the remote end. That new socket from the accept is the socket that you then use for send/recv.

Normally if you're a server, you would do a recv to get the request from the remote end.

roger-, 2015-07-03:
Doh, totally forgot this wasn't client only!

Thanks, I get it now.

closed 2015-07-03 by roger-