micropython/micropython #1238: CC3200, message arm-none-eabi-gcc: error: nano.specs: No such file or directory

asez73, 2015-05-07:
while making CC3200 bootloader and application, I have a compile time error with message arm-none-eabi-gcc: error: nano.specs: No such file or directory
File ../../localconfig.mk is invoqued in CC3200/Makefile but not provided in repository.
Looks like some local configurations are not set.
I do have an arm-none-eabi-gcc that can be accessed from command prompt...

dpgeorge, 2015-05-07:
nano.specs should come with your arm-none-eabi-* install.  Eg, for me I get:
```bash
% find /usr/arm-none-eabi -name nano.specs
/usr/arm-none-eabi/lib/thumb/nano.specs
/usr/arm-none-eabi/lib/armv7e-m/nano.specs
/usr/arm-none-eabi/lib/armv7e-m/softfp/nano.specs
/usr/arm-none-eabi/lib/armv7e-m/fpu/nano.specs
/usr/arm-none-eabi/lib/armv7-m/nano.specs
/usr/arm-none-eabi/lib/nano.specs
/usr/arm-none-eabi/lib/armv7-ar/thumb/nano.specs
/usr/arm-none-eabi/lib/armv7-ar/thumb/softfp/nano.specs
/usr/arm-none-eabi/lib/armv7-ar/thumb/fpu/nano.specs
/usr/arm-none-eabi/lib/armv6-m/nano.specs
/usr/arm-none-eabi/lib/fpu/nano.specs
```

`../../localconfig.mk` is not needed, it's only if you want to override defaults in the makefile.

danicampora, 2015-05-07:
Yes, exactly that's why is included as `-include`, it's only optional. It builds perfectly on Travis...

asez73, 2015-05-07:
Ok, I found some strange things in my configuration: apparently, there are 4 differents arm-none-eabi installed: TI, ARM and a stranger one scattered in differents sudirectories definitly not standard...
Sorry for this false bug report!
You should close it now.

closed 2015-05-07 by asez73