micropython/micropython #1240: [wiki] add list of FAQ

cloudformdesign, 2015-05-07:
@dpgeorge I think you would be the most qualified, when you get a quiet moment, to write down a few of the FAQ and responses that you can think of. I have created a page and linked it on the sidebar, but I have not been in the project long enough to give any good answers (or questions!)

https://github.com/micropython/micropython/wiki/FAQ

pfalcon, 2015-05-09:
I removed an over-optimistic (giving false hopes?) context-less statement of 10x more fastness than CPython. @cloudformdesign : Don't hesitate to do your research and testing to answer those questions, while waiting when other maintainers get to it, that's actually useful to have someone else to (re)validate results/process.

cloudformdesign, 2015-05-09:
Very sorry about that,  I could have sworn I read that from Damien somewhere.  I should have researched

pfalcon, 2015-05-10:
Well, you could be 100x slower on average, and still be "up to 10x faster". Saying only the latter part is hardly helpful to anyone, and will only raise credibility concerns in people who understand that it's not possible to just sit and write (without big efforts) an interpreter which is 10x times faster than another, long- and well-developed one. And those are exactly people we're interested in contributing.

pfalcon, 2015-05-15:
I've written a "performance" section for faq - went a bit long, but the tells everything. Add link to @dpgeorge's chart from https://github.com/micropython/micropython/issues/941 too.

Well, bottom line: we may need a TOC for FAQ ;-).

cloudformdesign, 2015-05-15:
that looks great, thanks for adding it!

Bringing up assembler made me realize that eventually it would be good to add a link on how to do C support. Cython (#658) would be helpful, as well as a wiki page documenting the differences between the CPython C api and micropython's.

pfalcon, 2015-08-25:
FAQ is filled in now (but whoever adds do it in the future should also think about adding answers, not just questions ;-) ).

closed 2015-08-25 by pfalcon