micropython/micropython #1246: Improve internal hash dispatch

dpgeorge, 2015-05-09:
Commit 37c6555b4419eb6027a345ee0c3e759d50da771f made changes to the way hash is handled for user types and non-builtin native types, and lead to a discussion: https://github.com/micropython/micropython/commit/37c6555b4419eb6027a345ee0c3e759d50da771f#commitcomment-11120008 .  Discussion raises important points about how hashing is dispatched internally so making it an issue for easier reference.

dpgeorge, 2015-05-09:
Note that CPython internally truncates the result from a user defined `__hash__` method to the machine word size, so we are also free to do this.

dpgeorge, 2015-05-09:
One suggestion is to add MP_UNARY_OP_HASH as a new unary op for the type->unary_op method.  The existing mp_obj_hash function returns an mp_int_t (not an integer wrapped up as an mp_obj_t), and so unary_op(MP_UNARY_OP_HASH, obj) should also return an mp_int_t instead of an mp_obj_t.  This is doable, it just requires some casting.

But the following problem arises: how to distinguish "op not supported", which is indicated by a return value of MP_OBJ_NULL (=0) from unary_op, and the return of a hash value of 0?

Hash values of zero are allowed:
```python
hash(0) # gives 0
hash(False) # gives 0
class A:
    def __hash__(self):
        return 0 # allowed
hash(A()) # gives 0
```

So it seems the only way to force unary_op to handle hashing is to require it to return the hash value packed into a small int.  But this seems inefficient: one must now pack and unpack the small-int value.  Furthermore, the hash value now loses a bit.

pfalcon, 2015-05-09:
> So it seems the only way to force unary_op to handle hashing is to require it to return the hash value packed into a small int. 

Yes, like OP_LEN.

>  But this seems inefficient: one must now pack and unpack the small-int value.

Well, it's just shift + or, calling itself is more inefficient.

> Furthermore, the hash value now loses a bit.

Not a big loss ;-).

dpgeorge, 2015-05-09:
Just seems like a lot of changes (and code increase) to fix something that was fine to begin with.

pfalcon, 2015-05-09:
> that was fine to begin with.

It wasn't exactly fine, it did linear search, it was inconsistent with how other unary operations implemented. Ok, if you think it makes more sense to write comments why the inconsistency and have duplication of looking up ``__hash__`` in user instances vs other native types (or extra conditions trying to reuse lookup for both of them), then well, that can be done. But I don't understand how's that more elegant then requiring that native types return non-0 hash from unary_op (e.g. by doing "return val | 1").

dpgeorge, 2015-05-10:
> it did linear search

On the most unused case (with exactly zero uses in core).

> it was inconsistent with how other unary operations implemented

But completely consistent within itself; one function to do it all.

> requiring that native types return non-0 hash from unary_op

Because hash(False) must be zero.

pfalcon, 2015-05-11:
> On the most unused case (with exactly zero uses in core).

There's such notion as structural and algorithmic beauty. Eventually, that bleeds thru to actual code too (even if currently there're "zero uses").

> But completely consistent within itself; one function to do it all.

Situation when each thing is consistent with itself would be nice scientific definition of "mess" ;-).

>>    requiring that native types return non-0 hash from unary_op
> Because hash(False) must be zero.

But emphasis there was on "why you think it's better to do with extra if's in mp_obj_hash rather than in unary_op?".

Anyway, let's come forward with it - decide how it will be and who does what.

dpgeorge, 2015-05-11:
My underlying concern with all this is that over time we will get code-size creep from changes that are not even making it more Python compliant.  We already have creep from changes to add needed features.  And on the other side, a lot of work goes into reducing code size, sometimes at the expense of code elegance.

I have a work in progress on this issue.

pfalcon, 2015-05-11:
Well, to objectively resolve this, we'd need to get both implementations, and compare look of the code and difference in binary size. That's arguably waste of resources, and *this* case is not that important after all.

But my concern is about eventually ending up with code like that:

~~~~
case H_(0ab31997,2d5213fa): case H_(bf875611,200e9990):  /* vector_size */
~~~~

I wanted to post that code in the previous comment, but had a look and thought: "but is that it really that bad to serve as anti-pattern"? Such thoughts are already a code elegance/clarity creep. So, doubts off, the code above is *bad* (unless it's auto-generarated, but there's no sign of it, and there're better ways to autogenerate it (we use that better way)). The code is from LuaJIT fro reference.

Anyway, thanks for working on this, I guess we can call discussion done then ;-).

dpgeorge, 2015-05-11:
I'll get some options for us to compare.  But you are right that it's not a good use of time.

dpgeorge, 2015-05-14:
Resolved.

closed 2015-05-14 by dpgeorge