micropython/micropython #1252: Allow qstrdefs to be preprocessed safely without macro name clashes

dpgeorge, 2015-05-11:
On Mac qstrdefs preprocessing is now broken; see https://github.com/micropython/micropython/commit/1129de5ac09ac5b41a952021cbc7a9752c8c253f#commitcomment-11139851

The problem is that some qstr names (in this case stdin, stdout, stderr) are #define'd to something else in a system header.  And so `Q(stdin)` is preprocessed to, eg, `Q(__stdinp)`.

[@pfalcon I think this has arisen due to 8fbabab1a80efa8b9c0654f63b2157d8f8299955.]

We need the preprocessor to run on qstrdefs.h to selectively include certain qstrs.  We need to include mpconfigport.h in qstrdefs.h to work out what to enable.  And mpconfigport.h can include system headers.

Currently qstrdefsport.h is not preprocessed (I think for this same reason of macro name clashes) and it would be nice to be able to pass that file also through the C preprocessor.

pfalcon, 2015-05-11:
> [@pfalcon I think this has arisen due to 8fbabab.]

Ok, let's revert it.

Applying proposed #undef's is ok workaround too IMHO.

dpgeorge, 2015-05-11:
> Applying proposed #undef's is ok workaround too IMHO.

No.  I've hit this before with other qstr ids, and we can't undef everything just in case it's defined somewhere.

blmorris, 2015-05-12:
I just tried to build the unix port on my MacBook with and without 8fbabab, I can confirm this is what caused the problem referenced at the top - but I'm pretty sure you both already knew that.

dpgeorge, 2015-05-30:
Now hitting this issue with the qstrs for `LITTLE_ENDIAN` and `BIG_ENDIAN` (they are being redefined to 1234 and 4321 by CPP).

dpgeorge, 2015-05-30:
Fixed.

closed 2015-05-30 by dpgeorge