micropython/micropython #1253: unix: build broken on OSX by requirement for GNU readline

blmorris, 2015-05-12:
(Breaking discussion from https://github.com/micropython/micropython/commit/1129de5ac09ac5b41a952021cbc7a9752c8c253f#commitcomment-11139851 now that it has been confirmed as a real issue.)

This is more of an issue with OSX than with MicroPython, but now that it has been confirmed by other OSX users we should open the discussion for possible solutions.
Prior to #1252 the unix build failed on OSX with the following message:
```
CC ../py/../extmod/modubinascii.c
CC ../py/../extmod/modmachine.c
CC main.c
CC gccollect.c
CC input.c
input.c:38:10: fatal error: 'readline/tilde.h' file not found
#include <readline/tilde.h>
         ^
1 error generated.
make: *** [build/input.o] Error 1
```
The issue is that the default readline on OSX is not GNU readline, but as mentioned in the prior discussion it is based on libedit from NetBSD. It is possible to use a package manager like MacPorts or HomeBrew to install GNU readline, however these package managers do not set the path to point to the new readline, presumably to avoid breaking software that relies on the native OSX version.
Inspired by a suggestion from @Anton-2, I found that I can build without modifying `unix/mpconfigport.h` with this command: `CFLAGS_EXTRA=-I/opt/local/include make -j10`
Presumably any other OSX user can do something similar, but it is less ideal than just being able to type `make -B` every time I pull the current source.
There was also the suggestion of linking against the package [gnureadline](https://pypi.python.org/pypi/gnureadline) like CPython apparently does to get around this issue on OSX, but I don't know how that might be done.
Opening this ticket to discuss these or any better suggestions.

Anton-2, 2015-05-12:
Well, on my version of OS X (10.10.3), the system readline (libedit) seems to be sufficient for micropython.
The problem comes from the missing header file ( readline/tilde.h ). This file does not exist, but the functions ( tilde_expand, ...) are defined in realine/readline.

So, by just commenting out
```c
#include <readline/tilde.h>
```
in input.c, I can compile micropython, with the system readline. And history works ;-)

Could you test on your version of OS X ? If it works then maybe we could just skip #include \<readline/tilde.h> on darwin, and use the system readline.

pfalcon, 2015-05-12:
So, to remind history of the issue, linking with readline is actually kind of problem, because it renders entire binary GPL with all consequences. Interested parties were invited to propose a (provable, not random) replacement for it. And nowadays, the best solution looks like switching to micropython's own implementation of readline (as used by bare-metal ports), now that it supports completion. And readline history support was kinda enabled to fix that as another requirement. As soon as builtin readline support configurable history depth and is able to save/load it, we can switch.

blmorris, 2015-05-12:
I can confirm that @Anton-2's solution works my system (10.7.5), I can compile micropython simply by commenting out `include <readline/tilde.h>` from input.c
So I guess there is another solution to avoid linking against GPL libraries - just use libedit instead ;-)
Although adding readline history to baremetal ports sounds good too...

blmorris, 2015-05-27:
Closing issue; fixed by #1286

closed 2015-05-27 by blmorris