micropython/micropython #1254: ESP8266 build broken by this afternoon's changes

owens-bill, 2015-05-13:
Since picking up the latest round of ESP8266 changes, the firmware no longer builds for me, instead throwing this error:

xtensa-lx106-elf-ld: build/firmware.elf section `.text' will not fit in region `iram1_0_seg'
xtensa-lx106-elf-ld: region `iram1_0_seg' overflowed by 156 bytes

dpgeorge, 2015-05-13:
It's fine for me (using version xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2 and GNU ld (crosstool-NG 1.20.0) 2.24).

Are you using the standard makefile?

owens-bill, 2015-05-13:
I am using the standard everything ;) I just updated esp-open-sdk, it's showing the same versions you are. I'm on Xubuntu 14.04.2 64bit, fairly clean and simple install, which up until yesterday was building the esp8266 code just fine. I've done the same test on a VM that was also working previously, and it now has the same error.

dpgeorge, 2015-05-13:
Ok I can confirm the problem.  I was using an older version of the ESP SDK.  With the latest version (1.0.1) indeed the iram1_0_seg overflows by 156 bytes.

I moved some more object files to the irom_0_0seg segment.  Now it should work.

closed 2015-05-13 by dpgeorge

owens-bill, 2015-05-13:
Builds fine for me now, thanks!