micropython/micropython #1260: stmhal: would like to switch off cooked mode for sys.stdout

dhylands, 2015-05-13:
Currently, when calling to sys.stdout.write, it eventually winds up calling mp_hal_stdout_tx_strn_cooked
https://github.com/micropython/micropython/blob/master/stmhal/pybstdio.c#L76

I'm trying to write a file transfer program which will transfer files using the raw repl. Basically I upload a small program which reads the file from the pyboards file system and then send the contents back. However, any 0x0A characters in the file are being translated into 0x0D0A sequences.

My converse program (for sending files to the pyboard) is having 0x0D converted into 0x0A's on the pyboard.

I have a number of options available to achieve this:

1 - be able to set some type of raw (or transparent) mode which will change the way stdio_read and stdio_write work.
2 - I could encode my data being sent back (something like base64)
3 - I could escape all 0x0Ds 0x0As and send them as something else

Ideally, I'd like to do 1.

But I figured I'd ask to find out if there was a preferred way.

pfalcon, 2015-05-13:
By definition, sys.stdin/stdout/stderr in Python are text-mode, thus you cannot rely on them being binary-safe. There're generally 3 ways out of that:

1. Use specific UART/USB_VCP objects which should work in binary mode (the best option apparently).
2. Have os.open/read/write() - too bloated for stmhal apparently, but what I use in e.g. pyedit.
3. To support CPython idioms which nobody knows: http://bugs.python.org/msg77230 - again, too bloated for stmhal, though I hope to support them in unix port (even though it's moot point as "nobody knows them", but I hope to find some existing code which does that as an excuse).

dpgeorge, 2015-05-13:
USB_VCP is in binary mode, and does not do any translation.  Is there a reason that you need to use sys.stdout?

We could add a termios-like function to set raw/cooked mode on stdio.  That wouldn't be pretty minimal.

> To support CPython idioms which nobody knows (ie sys.stdout.buffer).

I know!  And I've used it in uPy here: https://github.com/micropython/micropython/blob/master/tools/pyboard.py#L37

dhylands, 2015-05-13:
@dpgeorge I wanted to use stdin/stdout so that it works regardless of whether its connected over USB or over a serial link (say via bluetooth). Otherwise the host has to know if this is a usb link or a serial link and then generate different code.

The sys.stdin.buffer isn't present in uPy (the code you referenced is actually CPython code)

dpgeorge, 2015-05-13:
Ok.  If we did go for sys.stdout.buffer support then would that solve the problem?  Ie is sys.stdout.buffer in CPython supposed to be completely raw and uncooked?

It's actually not too much bloat: just a new IO type that is binary instead of text, and does no cooking for read/write.  All other methods can be reused.

dpgeorge, 2015-05-13:
See PR #1261 for proof-of-concept of buffer support (which I think is a very clean way to handle this).

Other option would be to multiplex it yourself over UART and USB_VCP, something like:
```
def write(data):
    uart.write(data)
    usb_vcp.write(data)
def read(data):
    if uart.any():
        return uart.read()
    else:
        return usb_vcp.read()
```

dhylands, 2015-05-14:
> Ok. If we did go for sys.stdout.buffer support then would that solve the problem? Ie is sys.stdout.buffer in CPython supposed to be completely raw and uncooked?

I think that whether something is raw or cooked is outside of the scope of python itself. In linux, it's controlled used termios, which basically reaches into the terminal driver. I don't think Windows or Mac even have notions of terminal drivers or line disciplines.

In micropython we're basically combining the python stuff with the os-level stuff. So I think addding a micropython specific method is also "acceptable".

Another option is to have a "pluggable" repl. If such a thing existed then you could replace the existing repl with one that did raw mode.

The only place in mcropython that we currently have anything cooked relates to the repl. Using raw mode with the REPL seems like a fairly rare case, so I'm happy with whatever solution is chosen (which might even be that we don't make any changes). I'm just trying to pick my direction.

dhylands, 2015-05-14:
After having read all of the comments and followed all of the links, I think I kind of like using PR #1261. Let me look at what might be involved with that and I'll do some testing/adjustments etc.

dhylands, 2015-05-14:
So I tried out PR #1260 and it seems to work fine for me.

I also hacked up a raw function, which does the same thing, only takes 88 bytes, but perhaps isn't as elegant PR #1262

dpgeorge, 2015-05-14:
> I think that whether something is raw or cooked is outside of the scope of python itself.

Right.  Except that, as you say, termios module can be used to control the raw/cooked state.

> Use specific UART/USB_VCP objects which should work in binary mode (the best option apparently)

This is not a good option because it's not portable across boards.  On the WiPy stdio is connected to the telnet server (and an optional uart) and there is no way to get a telnet file-like object to read/write data.   On ESP8266 there is no uart object (yet) and stdio is directly connect to the uart hardware.

It is therefore more general to use sys.stdin/stdout because they are more likely to exist on a given port (they even exist on pic16bit!).  And when writing some code that acts like a unix utility and reads/writes data it's just natural to use stdio rather than some board-specific object like UART or USB_VCP.

So I think we are down to deciding which of #1261 and #1262 are better.  Both of them work for all ports that use stmhal/pybstdio.c (which is stmhal, esp8266, cc3200, pic16bit, teensy).  #1261 (buffer) is compatible with CPython.

pfalcon, 2015-05-14:
> it's not portable across boards

Shame on those boards! And that exactly shows a problem, why I criticize adding non-standard stuff: we have a bunch of it already, but people don't know about it, and happy to add even more non-standard - usually completely adhoc and for a single port they care at the moment. (With a disclaimer that I myself added a bunch non-standard stuff, and happy to add more, but - good news - I restrain myself on the latter, making sure I prioritize adding *standard* stuff first).

>  #1261 (buffer) is compatible with CPython, while #1262 is not.

Per above, I think we should follow standard Python way inasmuch as possible, unless there're *really* good reasons to do otherwise.

danicampora, 2015-05-14:
> there is no way to get a telnet file-like object to read/write data

There might be a way if this is really necessary. I try to keep things minimal for the CC3200 and add/support features as they become needed.

> Per above, I think we should follow standard Python way inasmuch as possible, unless there're really good reasons to do otherwise.

+1. We should go the standard way whenever is possible and the path is clear.

dpgeorge, 2015-05-14:
> And that exactly shows a problem, why I criticize adding non-standard stuff

This is not a valid example to criticise that.  Even if we had a perfect standard we would still have problems: eg pic16bit doesn't have room for UART, but does provide sys.stdio.  Eg cc3200 doesn't need or want to provide telnet file-like object (how would it do it in a standard way?), exactly because there's sys.stdio for it.

And adding sys.stdio.buffer is exactly trying to provide something that will be portable and CPython compatible across all boards, and even unix port.

dhylands, 2015-05-14:
What if make it configable and have it on by default? (at least for the ports that aren't starved for space).

Personally, I still favor #1261 (bufffer) even though it takes more flash. My end goal is to have enough file transfer capability to make it possible to turn off mass storage mode.

I only created #1262 as a discussion point.

dpgeorge, 2015-05-24:
sys.stdXX.buffer now implemented in 3c4b5d42817944438c16fe6f1c70447ffc751110.

closed 2015-05-24 by dpgeorge