micropython/micropython #1268: Micro Python v1.4.3 slower than v1.4.2 by factor 5 !

fojie, 2015-05-17:
Hi,
Being an amateur, I still hope I can contribute:

I found that the Micro Python v1.4.3 release for pyboard is much slower than 1.4.2 and v1.4.2-50-ga9afcb1. I have a program that writes on a TFT graphic display using SPI (10.5 Mbaud) and my test program takes 5 times longer with the latest version v1.4.3. (I have several pyboards and always use the version including drivers for cc3000).

I grabbed some pictures from a logic analyzer showing the exact same sequence. See the extract of the method in question and pictures below: Two Pins are shown in the script and graphs (D/C* and Debug used for trigger). These gives a good measure of timing. (The SPI signals are way under sampled).

I you find this useful, don't hesitate to contact me for more info. (folke (at) me.com)

class St7735:
	# Constants
	DC      = "Y4"
	CS_TFT  = "Y5"
	CS_SD   = "Y3"
	LITE    = "Y2"
	RST     = "Y1"

	def __init__(self):
		self.dbg = Pin("X9", Pin.OUT_PP)        # Debug pin
		self.dbg.low()
		self.spi = SPI(2, SPI.MASTER, baudrate=10500000)
		ss = Pin(self.CS_TFT, Pin.OUT_PP)
		ss.low()                                # enable TFT ALWAYS!
		self.dc = Pin(self.DC, Pin.OUT_PP)      # Data/Command*
		self.dc.low()
		self.DC_flag = False
		cs_sd = Pin(self.CS_SD, Pin.OUT_PP)     # SD-reader enable*
		cs_sd.high()                            # Disable SD-reader ALWAYS!
		self.lite = Pin(self.LITE, Pin.OUT_PP)
		self.lite.high()                        # TFT backlite on
		reset = Pin(self.RST, Pin.OUT_PP)
		reset.low()                             # Reset Display
		delay(2)
		reset.high()
		delay(200)                              # Should be enough

	def FastDrawPixel(self, x, y, color):
		""" Draws a pixel on the TFT display; increases speed by inlining """
		self.dbg.high()		####### Debug set HIGH  #######
		self.dc.low()           ####  # set D/C line to 0 = command
		self.spi.send(self.CA_SET)
		self.dc.high()          ####  # Set  D/C line to 1 = data
		self.spi.send(bytearray([0, x, 0, x]))
		self.dc.low()           ####  # set D/C line to 0 = command
		self.spi.send(self.RA_SET)
		self.dc.high()          ####  # Set  D/C line to 1 = data
		self.spi.send(bytearray([0, y, 0, y]))
		self.dc.low()           ####  # set D/C line to 0 = command
		self.spi.send(self.RAM_WR)
		self.dc.high()          ####  # Set  D/C line to 1 = data
		self.DC_flag = True
		self.spi.send(bytearray([color >> 8, color & 0xFF]))
		self.dbg.low()		####### Debug set LOW ########

[image]
[image]

dpgeorge, 2015-05-17:
Thanks for the report.  It's strange because not much has changed that would affect performance on the pyboard.  In fact, running the Pystone(1.2) benchmark on the pyboard I actually see an increase in performance from v1.4.2 to v1.4.3: 1740 pystones up to 1830!

pfalcon, 2015-05-17:
@fojie: We'd need more info, like the actual code to run and reproduce the problem - without a scope, and preferrably on unix port (which may be hard in this case, but otherwise is a good measure of reproducibility). Then, where you got your 1.4.2, v1.4.2-50-ga9afcb1 and v1.4.3 from (in a report without much specific detail, like this, first suspicion falls on how different versions were compiled).

fojie, 2015-05-17:
@pfalcon: I fetch all dfu files from the official "Micro Python downloads". I still have the v1.4.2-50-ga9afcb1 from a previous download, so I could go back (without loosing recognition). It's a pity that the nightly builds of v1.4.2 are removed, otherwise I could quickly nail down which of them introduces the problem. Or could I find them somewhere else ? (I assume it would be helpful to find the first with the problem?)

Otherwise I will work some more to try to conclude if:

A/  Is the momentaneous SPI bitrate lowered?
B/  Is the time between SPI transfers longer?
C/  Are Pin methods slower?
D/  Is it maybe something else than the SPI ?

I have the feeling that it's not necessary to have my (rather big) program to reproduce the problem. Yes, it sends a lot of SPI data and also controls some extra Pins, but that's easy to implement.

dhylands, 2015-05-17:
Here's a summary of all of the commits since a9afcb1 https://pastebin.mozilla.org/8833767

If you can build the firmware yourself, you can go back in time and recreate any of them.

None of them jump out at me (except maybe the hash stuff) as being culprits.

dpgeorge, 2015-05-17:
> Or could I find them somewhere else ? (I assume it would be helpful to find the first with the problem?)

I've made available all the versions between the good and bad ones, see http://micropython.org/resources/firmware2

@fojie are you able to find out which version introduces the problem?  It's best if you use binary search, ie try the middle one first and see if it fails/passes then choose the next one to test based on the outcome.  Also, successive firmware versions with the same size should be exactly the same so no need to try those.

Thanks!

fojie, 2015-05-17:
@dhylands: Thanks. But I should have been clear: I am not able to build myself, so I need .dfu files ready to be installed via df-util (on Mac OS X).

@dpgeorge: Thanks, I'll try.

I have looked further with the Logic Analyzer and it seems that there is no problem with Pin, but a lot of dead time between the bytes on MOSI when sending multiple data bytes (spi.send(bytearray(4 databytes)) compared to sending only one data byte. But this is preliminary, I have to check more.

fojie, 2015-05-17:
@dpgeorge: Well "pybv10-network-v1.4.2-65-ge04aa96.dfu" is slow while "pybv10-network-v1.4.2-60-g04ee598.dfu" is fast. (Not tested those in between, since they are same sizes as the tested ones).

dpgeorge, 2015-05-17:
@fojie thanks... but that just makes it stranger... can you please try the ones in between 60 and 65 to find the exact commit that goes bad.

Also, are you able to run your code without the CC3000 driver?  It could be that there is some strange "interference" with the network driver and the SPI.  If you can then I'll make dfu images without the network drivers for you to test.

fojie, 2015-05-17:
@dpgeorge: OK, I'll do that…

My (latest) theory is that it has to do with the baud rate; if I use 10.5 Mbaud it's SLOWER on the slow builds than using 2 Mbaud as I try now. (I want to be able to catch clock and data on my simple Logic Analyzer; More checking to do)

I also sometimes get OS error 5 now (is that TimeOut eror?). I find it odd since MISO is not used, although connected to the TFT controller (I never read from the TFT). Earlier I thought thats because i tried to run at 21 Mbaud (and even 42 ?), but AFAIK there is no handshaking on SPI!?

dpgeorge, 2015-05-17:
@fojie Note that the SPI baudrate that you ask for is not always what you get.  There are only set rates you can do (it depends on the bus frequency, which yau can change using pyb.freq).  To check the rate, print out the SPI object after it's initialised.

There have also been reports by others about one-way SPI not working properly.  Try setting the MISO pin to a standard GPIO pin using pyb.Pin(MISO, pyb.Pin.IN); that might fix it.

fojie, 2015-05-17:
@dpgeorge: I knew about the baud rate and prescaler, that's why I use 10.5 MHz. I have tested the MISO as GPIO-Pin. (I assume it should be setup after creating the SPI object). I still get occasional OS Error 5 and no change in speed. Was this fix supposed to fix OS error 5?

I have tested #59–63 also now, and guess what: 62 is OK 63 is slowish; at first 'in between' – after repeating – slow. So there is probably some temperature influence. Immediately after firmware update and reset it's a little (≈ factor 2) faster than after soft reboot. So this is probably not easy to find. Maybe that's why I also earlier saw that lower baud rate improved speed. So there is a risk that, me saying this is fast and that is slow, is not so correct after all due to things like temperature…

I can just add that my 13 year old son is learning Python on his pyboard with an identical TFT. Same factor 5 slower on his system with v1.4.3, (same graphical driver).
 and I used 2 other pboards during theese tests.

I sure can test without cc3000 (I got it working ONCE only anyway!), so if you want me to check w/o network drivers, feel free to make them available. But I will not test more tonight. 

I want to thank you guys for helping me so far.
/Folke Berglund, Sweden

fojie, 2015-05-18:
@dpgeorge: Today I tested to use the NON-network version v1.4.3 (pybv10-2015-05-17-v1.4.3.dfu) from the download page. Result: it's *fast* !! So maybe my problem has something to do with the cc3000 driver? If I by pure interest (if it's buffered, use DMA, IRQ…) want to look at the SPI driver, what source module(s) contains it?

Since this is my first issue, I want to know if it's ok to leave this open for a while? (I don't know if that really is supposed to be my decision. I'm totally knew to Github and just joined and got fascinated by the incredible work you guys do.

dhylands, 2015-05-18:
@fojie You'll find the python interface to SPI here:
https://github.com/micropython/micropython/blob/master/stmhal/spi.c

and the lower level implementation here:
https://github.com/micropython/micropython/blob/master/stmhal/hal/src/stm32f4xx_hal_spi.c

and the cc3000 driver here:
https://github.com/micropython/micropython/tree/master/drivers/cc3000/src and the spi stuff for the cc3000 driver seems to be mostly in ccspi.c

Leaving the issue open for a while should be fine.

fojie, 2015-05-18:
@dhylands Thank you! Now I will have no problem with what to do on my spare time.

A little off topic: Dave, i have not seen anything about the cc3000 for a while. Have people given up with it or is it functioning? I have two of them with a somewhat old firmware. Earlier I saw people discussing how to get them running with the pyboard, but now it seems quiet. Are people waiting for the WiPy (as I am), or does the cc3k function?

dhylands, 2015-05-18:
@fojie I'm not sure - I haven't used the cc3000 myself yet. I believe that some people are using it, but I've heard of problems associated with it. I have signed up for the WiPy, and by the time I'm ready to play with Wifi, I expect I'll have my WiPy.

dpgeorge, 2015-05-21:
@fojie Do you have the cc3000 driver initialised when you get the problem of slow SPI?  Do you create a CC3K instance, or do you just have the SPI code?

fojie, 2015-05-22:
Now I have made a small test program that clearly shows the problem: it test all available baud rates with both "graphical" data (as when I hit the problem), as well as generic of two kinds – one similar to graphical data but with no sign of problem. The other with the problem. The latter has the problem when it sends 3 or 4 bytes in each spi.send (not for 2 bytes). But the problem only(?)  shows when data is prepared on the fly, NOT if it is pre-prepaired into a long buffer. So there could be some kind of **race** condition in the SPI driver, (between calling and DMA/IRQ ?)
The program can be run on a naked pyboard and here is *part of* a printout with Network build 1.4.2 63:

@dpgeorge: I just have the SPI code (as attached below) on a naked pyboard! [Honestly, I have given up on the cc3k, since it hangs for ever on init (on both my copies, maybe their firmware is too old).]

Micro Python v1.4.2-63-g3d3ef36 on 2015-05-17; PYBv1.0 with STM32F405RG
NETWORK build:
Graphical type data (8192 bytes):
Sometimes slow @baud:   0.328 Mbit/s    0.290 s
Sometimes slow @baud:   0.656 Mbit/s    0.193 s
Sometimes slow @baud:   1.313 Mbit/s    0.143 s
Sometimes slow @baud:   2.625 Mbit/s    0.117 s
Sometimes slow @baud:   5.250 Mbit/s    0.106 s
Sometimes slow @baud:  **10.500 Mbit/s    2.049 s  !!!!!!!**
Sometimes slow @baud:  21.000 Mbit/s    0.101 s

Generic data (8192 bytes **pre-prepared**):
Always fast @baud:   0.328 Mbit/s    0.200 s
Always fast @baud:   0.656 Mbit/s    0.100 s
Always fast @baud:   1.313 Mbit/s    0.050 s
Always fast @baud:   2.625 Mbit/s    0.025 s
Always fast @baud:   5.250 Mbit/s    0.013 s
Always fast @baud:  10.500 Mbit/s    0.006 s
Always fast @baud:  21.000 Mbit/s    0.003 s
_______________________________________________________________
Test program:
"""Test speed of SPI port for different baudrates and data"""

from pyb import SPI, Pin, delay, micros, elapsed_micros, rng

def print_elapsed_time(function, baud, start_time):
	""" Formats an output string showing elapsed time since doing 'start_time = micros()' """
	print("%15s: %7.3f Mbit/s %8.3f s" % (function, baud/1e6, elapsed_micros(start_time) / 1e6))

baud_rates = [328125, 656250, 1312500, 2625000, 5250000, 10500000, 21000000]
spi = SPI(2, SPI.MASTER, baud_rates[0])

""" Test SPI with data similar to graphical TFT traffic; Slow for some baudrate and newer Network builds """
print("\nGraphical type data (8192 bytes):")
for baud in baud_rates:
	spi.init(SPI.MASTER, baud)          # Change baud rate
	'''Since MISO is not connected, disable SPI for MISO by setting Pin as normal Pin input'''
	miso_dummy = Pin('Y7', Pin.IN)      # May help prpevent OS error 5 ?

	start = micros()

	for a in range(128):
		for x in range(16):
			spi.send(bytearray([0, x, 0, a]))

	print_elapsed_time('Sometimes slow @baud', baud, start)

""" Test SPI with generic data; Always(?) fast! """
print("\nGeneric data (8192 bytes pre-prepared):")
for baud in baud_rates:
	spi.init(SPI.MASTER, baud)          # Change baud rate
	data = bytearray([0, rng() & 0xFF, 0, rng() & 0xFF] * 2048)
	start = micros()
	spi.send(data)
	print_elapsed_time('Always fast @baud', baud, start)

""" Test SPI with simulated Graphical data; Slow for some baudrate and newer Network builds """
print("\nGeneric data similar to Graphical  (8192 bytes):")
for baud in baud_rates:
	spi.init(SPI.MASTER, baud)          # Change baud rate
	start = micros()
	for _ in range(2048):
		spi.send(bytearray([0, 0, 0, 0]))
	print_elapsed_time('Sometimes slow @baud', baud, start)
print()

fojie, 2015-05-23:
I have updated the test program (see previous comment), and done more extensive test runs. The **issue is detected in all tested versions**, *even the standard version w/o Network! Just more often in the newer version with Network*. So probably: the **issue is present in all versions**.

It's clear(?) that at some combinations of baud rate and number of sent bytes, cause some kind of **race condition**. My *guess* it is in this situation: calling spi.send with new data when previous data finished and thus (I assume) the DMA raises IRQ at a bad time since the DMA should be set up for the new data (or something like that). What seems to happen is some kind of **time out for 1 ms**, which is clearly visible in the print out from the test program. Time between calls increases up to 50-fold!

I have only found the issue at baud rates 21 and 10.5 Mbit/s, but since it's not the speed, but the timing that's causes the issue, it might happen also at lower baud rates when the loops (bursts) have the critical interval (seems to be ≈45–60+ µs depending of data size, but should be related to time of DMA end).

Two examples of parts of the print outs: (Sorry for the long text, but I feel it's needed).
<pre><code class="python">**Micro Python v1.4.3** on 2015-05-17; PYBv1.0 with STM32F405RG
**No Network**:
Graphical type data (2048 bursts with 4 bytes):
         0.328 Mbaud  Tot_time:  0.284 s  138.6 us between calls
         0.656 Mbaud  Tot_time:  0.188 s   91.9 us between calls
         1.313 Mbaud  Tot_time:  0.136 s   66.6 us between calls
         2.625 Mbaud  Tot_time:  0.115 s   56.4 us between calls
         5.250 Mbaud  Tot_time:  0.102 s   49.6 us between calls
        10.500 Mbaud  Tot_time:  0.096 s   46.9 us between calls
        21.000 Mbaud  Tot_time:  0.097 s   47.6 us between calls
................
Generic data (585 bursts with 14 bytes)
         0.328 Mbaud  Tot_time:  0.227 s  387.7 us between calls
         0.656 Mbaud  Tot_time:  0.129 s  220.7 us between calls
         1.313 Mbaud  Tot_time:  0.078 s  132.8 us between calls
         2.625 Mbaud  Tot_time:  0.053 s   90.3 us between calls
         5.250 Mbaud  Tot_time:  0.041 s   69.3 us between calls
        10.500 Mbaud  Tot_time:  0.037 s   63.0 us between calls
**SLOW !**  21.000 Mbaud  Tot_time:  0.583 s  997.0 us between calls
.
Generic data (546 bursts with 15 bytes)
         0.328 Mbaud  Tot_time:  0.225 s  413.0 us between calls
         0.656 Mbaud  Tot_time:  0.126 s  230.8 us between calls
         1.313 Mbaud  Tot_time:  0.078 s  143.0 us between calls
         2.625 Mbaud  Tot_time:  0.051 s   94.2 us between calls
         5.250 Mbaud  Tot_time:  0.039 s   71.7 us between calls
        10.500 Mbaud  Tot_time:  0.035 s   64.8 us between calls
**SLOW !**  21.000 Mbaud  Tot_time:  0.544 s  996.5 us between calls
.
Generic data (512 bursts with 16 bytes)
         0.328 Mbaud  Tot_time:  0.224 s  438.2 us between calls
         0.656 Mbaud  Tot_time:  0.125 s  243.9 us between calls
         1.313 Mbaud  Tot_time:  0.078 s  152.2 us between calls
         2.625 Mbaud  Tot_time:  0.050 s   98.1 us between calls
         5.250 Mbaud  Tot_time:  0.038 s   74.1 us between calls
        10.500 Mbaud  Tot_time:  0.032 s   63.3 us between calls
        21.000 Mbaud  Tot_time:  0.031 s   61.3 us between calls
############################

**Old Versions** also shows the issue – This **With Network**

**Micro Python v1.3.10-258**-gec1b1cf on 2015-03-25; PYBv1.0 with STM32F405RG
**With Network**
Graphical type data (2048 bursts with 4 bytes):
         0.328 Mbaud  Tot_time:  0.287 s  140.1 us between calls
         0.656 Mbaud  Tot_time:  0.191 s   93.5 us between calls
         1.313 Mbaud  Tot_time:  0.141 s   68.9 us between calls
         2.625 Mbaud  Tot_time:  0.115 s   56.2 us between calls
         5.250 Mbaud  Tot_time:  0.104 s   51.0 us between calls
**SLOW !**  10.500 Mbaud  Tot_time:  2.048 s 1000.2 us between calls
        21.000 Mbaud  Tot_time:  0.099 s   48.3 us between calls
.
Generic data (8192 bytes pre-prepared as ONE long buffer):  (**Always OK**)
         0.328 Mbaud  Tot_time:  0.200 s 199767.0 us between calls
         0.656 Mbaud  Tot_time:  0.100 s 99903.0 us between calls
         1.313 Mbaud  Tot_time:  0.050 s 49972.0 us between calls
         2.625 Mbaud  Tot_time:  0.025 s 25006.0 us between calls
         5.250 Mbaud  Tot_time:  0.013 s 12525.0 us between calls
        10.500 Mbaud  Tot_time:  0.006 s 6283.0 us between calls
        21.000 Mbaud  Tot_time:  0.003 s 3164.0 us between calls
.
Generic data (8192 bursts with  1 bytes)
         0.328 Mbaud  Tot_time:  0.516 s   63.0 us between calls
[... 44 lines cut ...]
         0.328 Mbaud  Tot_time:  0.227 s  388.9 us between calls
         0.656 Mbaud  Tot_time:  0.128 s  219.1 us between calls
         1.313 Mbaud  Tot_time:  0.078 s  134.0 us between calls
         2.625 Mbaud  Tot_time:  0.055 s   94.5 us between calls
         5.250 Mbaud  Tot_time:  0.041 s   70.4 us between calls
        10.500 Mbaud  Tot_time:  0.036 s   61.1 us between calls
**SLOW !**  21.000 Mbaud  Tot_time:  0.585 s  999.2 us between calls
.
Generic data (546 bursts with 15 bytes)
         0.328 Mbaud  Tot_time:  0.226 s  414.1 us between calls
         0.656 Mbaud  Tot_time:  0.127 s  232.1 us between calls
         1.313 Mbaud  Tot_time:  0.077 s  141.0 us between calls
         2.625 Mbaud  Tot_time:  0.054 s   98.6 us between calls
[... cut ...]
**SLOW !**  21.000 Mbaud  Tot_time:  0.546 s  999.9 us between calls
.
Generic data (512 bursts with 16 bytes)
         0.328 Mbaud  Tot_time:  0.225 s  439.4 us between calls
         0.656 Mbaud  Tot_time:  0.127 s  247.1 us between calls
         1.313 Mbaud  Tot_time:  0.076 s  147.9 us between calls
         2.625 Mbaud  Tot_time:  0.053 s  102.8 us between calls
         5.250 Mbaud  Tot_time:  0.039 s   75.2 us between calls
        10.500 Mbaud  Tot_time:  0.033 s   64.4 us between calls
        21.000 Mbaud  Tot_time:  0.030 s   59.0 us between calls

fojie, 2015-05-24:
Since I don't have the knowledge to fix this issue (I'll leave that to you guys, sorry), should I rename the issue to something more appropriate like: "SPI slow when sending bursts at certain intervals"? Suggestions ? Anything else I could help with?

dpgeorge, 2015-05-24:
@fojie Thanks heaps for the detailed info.  It looks like a DMA issue.  Note that the SPI driver uses DMA only if irqs are enabled, otherwise it uses polling.

Please, change the issue name to something you feel better reflects the problem.

dpgeorge, 2015-06-03:
I can confirm the problem on a non-network board.  It's slow for 14 byte bursts and 15 byte bursts at 21 Mbaud.

dpgeorge, 2015-06-03:
Ok, should be fixed by 53a8aeb6e7a89ad169ad3f9e5553d51fb86c03b8.

The problem was that the DMA busy-wait loop was using wfi, which meant that it had to wait for an IRQ before returning.  Usually the DMA completion IRQ fired and that made it return, but in some cases the DMA IRQ would fire before the wfi and hence the wfi needed to wait until the next sys-tick (1ms).  Simple fix was to remove the wfi.

@fojie could you please check the latest firmware (available in 2 hours) works for you?  Also, I might make your test script part of the pyboard test suite, is that ok?

fojie, 2015-06-03:
You guys are amazing! I was just scanning thru the source for the SPI and DMA, and I had to give up. Too much code for me!   @dpgeorge I'm honored if you can find any use for my script. I will be glad to test the new version, but I have to get some sleep before that.

I can't refrain from asking: Why do I need a dummy parameter (foo) in a callback definition:
 <pre><code class="python">def scan_timer_callback(foo):           # why is parameter foo needed?
called by:
tim.callback(scan_timer_callback)  # Timer 4 at 100Hz
</code></pre>If no foo: it will be called once only, instead of repetitive. (the code is not in a class)  **v1.4.3**

dpgeorge, 2015-06-03:
> Why do I need a dummy parameter (foo) in a callback definition:

It's not dummy, it's useful, it's the timer object itself.  Passing the timer that triggered the irq/callback to the handler is useful, eg, to stop the timer.  It also means you can use the same callback for multiple timers and have possibly different behaviour depending on which timer fired.

fojie, 2015-06-03:
Aha; smart! The system gives the argument, and the callback the parameter for free! So my my guess that's it not an 'issue', was correct. It's a feature. Thank you.

fojie, 2015-06-04:
OK, I can confirm that SPI works under **v1.4.3-110-g53a8aeb**, (versions with and w/o Network), on pyboard. I tested  the full speed range with *test_SPI* and the original graphical script where the issue was discovered at 10.5 Mbaud. Good work!

I guess I'm supposed to close the issue now?

dpgeorge, 2015-06-04:
Ok, great!  Yes @fojie you can have the honour of closing the issue :)  Thanks for the very well presented issue.

fojie, 2015-06-04:
@dpgeorge Oops! Sorry, I didn't know how it is supposed to be done.  (Next time maybe I know better...). But anyway, thanks for your patience!

closed 2015-06-04 by fojie