micropython/micropython #1269: Consistent ifconfig for ports with wifi

dpgeorge, 2015-05-17:
See https://github.com/micropython/micropython/pull/1267/files#r30475364 for background.
```
ifconfig((ip, subnet, gateway, dns)) sets a static IP.
ifconfig(()) with an empty tuple sets the IP mode to dynamic.
ifconfig() gets the interface config
```

dpgeorge, 2015-05-17:
The reason ifconfig takes and returns a tuple of the same structure is so that you can save the state and then restore it easily, eg:
```
conf = wlan.ifconfig()
...
wlan.ifconfig(conf)
```

An empty tuple for DHCP mode seems a bit obscure and someone who didn't know the API would be very confused what it meant.  How about something super obvious:
```
ifconfig('dhcp')
ifconfig('dynamic')
```
?

danicampora, 2015-05-18:
OK, I am happy with either:
```
ifconfig('dhcp')
ifconfig('dynamic')
```

Then, the next question is: ifoncfig as it is now in the CC3200 also returns the following info as a namedtuple:

    WiFi role (AP, station or P2P)
    SSID (own ssid if in AP mode or the one it is connected to)
    BSSID (MAC address of the AP it is connected to, or its own if in AP mode)
    MAC (its own MAC address)

Where to put this info now? Maybe `wlan.status()`? And keep it namedtuple?

danicampora, 2015-05-18:
What about `wlan.info()` to retrieve:

```
WiFi role (AP, station or P2P)
SSID (own ssid if in AP mode or the one it is connected to)
BSSID (MAC address of the AP it is connected to, or its own if in AP mode)
MAC (its own MAC address)
```

?

danicampora, 2015-05-19:
I have yet another idea for `WLAN.info()`:

`(mode, ssid, bssid, name, connections)`

`mode`: AP, Station or P2P
`ssid`: Our own SSID (if applicable)
`bssid`: Our own MAC address
`name`: Our own P2P name (if applicable, `None` otherwise)

Connections is a list of tuples with all the devices connected. In Station mode or P2P mode there's only one element in the list, but in AP mode the list will contain all the stations connected.

`connections` could be list of: `(ssid, bssid, name)`

`ssid`: SSID of the AP (if applicable, `None` otherwise)
`bssid` MAC address of the AP or the other P2P device
`name`: Name of the other P2P device. (if applicable, `None` otherwise)

Maybe this is too much information together, and the list of connections part could be moved to its own method, e.g. `WLAN.connections()`

roger-, 2015-05-19:
Do any of these boards let you use DHCP but override the DNS?

danicampora, 2015-05-19:
I don't think so.

dhylands, 2015-05-19:
Wouldn't you normally do that on the DHCP server side? Many DHCP servers allow customizations per mac address, so you can do pseudo static stuff, or change other attributes like gateway or DNS servers.

danicampora, 2015-05-20:
In https://github.com/micropython/micropython/commit/56053c37cf5ffc6cb81e67817d8e77c490ca7e38 I did the following:

`WLAN.ifconfig()` return and receives `(ip, subnet, gateway, dns)`, but if `'dhcp'` is passed instead it goes to auto ip mode.

`WLAN.info()` returns: `(mode, security, ssid or name, own mac)`

`WLAN.connections()` returns a list of: `(ssid or name, mac)`

I hope this is all OK ;-)

dpgeorge, 2015-05-20:
@danicampora I didn't yet get time to think about this properly, but my first guess would be that ifconfig should do everything: it should be able to set the parameters and also return all relevant info.  Using an attrtuple will mean that it can be made compatible with other network drivers that don't need/want to return all the info.  Ie at least ip, subnet, gw and dns are returned, but the rest are optional.

danicampora, 2015-05-20:
@dpgeorge That was my first implementation, but then you said that:
> The reason ifconfig takes and returns a tuple of the same structure is so that you can save the state and then restore it easily.

And it made me think that `(ip, subnet, gateway, dns)` is pretty neat and obvious and there's no need for an attrtuple there (which takes more code)...

danicampora, 2015-05-20:
`WLAN.connections()` as it is would be valid for Bluetooth as well.
`WLAN.info()` we could make it an attrtuple so that it also fits other network drivers.

I think keeping `ifconfig` as it is now makes it consistent in the sense that you can get and set all params through it. It looks less clean if we add mode, mac and other stuff that can only be retrieved.

dpgeorge, 2016-05-06:
WLAN interfaces should support ifconfig(), and config() for more specific parameters.  See ESP8266 port for reference implementation.

closed 2016-05-06 by dpgeorge