micropython/micropython #1275: Enhancement suggestion: gc.disable() and heap lock

peterhinch, 2015-05-21:
Would it be feasible, when gc.disable() is issued, to unlock the heap? Clearly it would need to be re-locked when gc.enable() was issued and for the duration of gc.collect(). The purpose would be to enable a user to write interrupt callbacks which used the heap. It would be their responsibility to disable the relevant interrupts before issuing gc.collect() or gc.enable().

Envisaged use case: developers of control systems wishing to run the principal control loop as a timer callback for a well-defined update rate. Many filtering algorithms depend on floating point operations.

pfalcon, 2015-05-21:
I suggest adding CPython behavior description as argumentation - if there it's possible to use heap after gc.disable(), then uPy is incompatible.

Otherwise, did you try to implement it, run the testsuite, then add more testcasesto trigger errors when gc.disable() active and see how it behaves?

peterhinch, 2015-05-22:
Before attempting a hack I'd appreciate some guidance from experienced devs who understand MicroPython memory allocation. If the idea has no chance of working it would save us time if someone told me at the outset.

dhylands, 2015-05-22:
The I see is that interrupts can happen at any time.
Even if we assume that you only garbage collect at specific times, you'd need to ensure that interrupts are disabled during those times, or make the garbage collector disable interrupts and re-enable them at strategic places where its ok to allocate while a garbage collection is happening.

You also need to consider that an allocation by an interrupt handler could be interrupting another allocation by the main thread, or even an allocation by another interrupt handler (interrupts can be nested).

My background is linux kernel, and you're not even allowed to use floating point anywhere in the kernel, let alone just an interrupt handler. I figure if the linux kernel can get away without any floating point at all, then microcontroller code can probably do the same.

My thoughts are that the interrupt handler should just act as a trigger and the real work should happen in the main thread. If you're running in real-time the main thread will be running frequently enough to do the work.

I agree that things would be easier if you could allocate during an interrupt handler, but I don't think its the end of the world that you can't.

peterhinch, 2015-05-22:
OK, thanks for that. In particular I hadn't appreciated that allocations are non-interruptable.

closed 2015-05-22 by peterhinch

dpgeorge, 2015-05-27:
> I suggest adding CPython behavior description as argumentation - if there it's possible to use heap after gc.disable(), then uPy is incompatible.

It's possible to allocate on the heap after calling gc.disable().  That call just disables auto collection, not heap allocation itself.

What's really needed is a gc.lock() and gc.unlock() pair of functions.  But as discussed above, using these in irqs is not possible because of nested allocation.

There was a PR long ago about making the GC reentrant, see #913, but it was not merged.