micropython/micropython #1280: gcc-5.x support

dpgeorge, 2015-05-23:
Arch Linux has upgraded to gcc-5.1.0 as the standard gcc version (previously 4.9.2) and so I'm now using this version to compile uPy.  Everything seems to work fine except for this patch 0d31bbc7fa3bdf10a14d92bb010a5016dc6101af which was needed due to more aggressive file-level optimisation.

There has been a significant decrease in Thumb2 code size with 5.1:
```
bare-arm -640
cc3200 -1104
stmhal -1964
```
Yes, down almost 2k on stmhal!  Unfortunately unix 32bit is up +129 bytes.

Change log for gcc is https://gcc.gnu.org/gcc-5/changes.html

One interesting thing I see is the support for integer overflow checking using `__builtin_add_overflow` and sub and mul.  See https://gcc.gnu.org/onlinedocs/gcc/Integer-Overflow-Builtins.html.  It would be nice to use this (optionally since other compilers might not support such operations).  But sadly I don't think we can benefit because our overflow checking must make sure the result fits in a small-int (which is smaller than the native int type).

Anyway, the point of this issue is more to advertise gcc-5 support than anything.

danicampora, 2015-05-23:
Awesome! those code size savings are more than welcome :-)

danicampora, 2015-05-24:
Maybe this is a stupid question... but, is there a way to install this version (5.1) of gcc-arm in Ubuntu?

dpgeorge, 2015-05-27:
@danicampora You can compile from source :)

gcc-5.1.x is now used to compile the daily binaries available at http://micropython.org/download.  You can see the small decrease in code size at http://micropython.org/resources/code-dashboard/

closed 2015-05-27 by dpgeorge

danicampora, 2015-06-17:
> @danicampora You can compile from source :)

I did it :-), thanks.