micropython/micropython #1282: Add enough support to unix version of micropython to run rshell

dhylands, 2015-05-24:
So now that I have rshell.py, which requires CPython 3.4 running on the host, it occurs to me that it would be interesting to be able to run rshell.py using micropython on the host.

Probably the most significant module that needs to be implemented is the serial module.

If we wanted this to work on other platforms (win/mac and unix) then I think that it makes sense to implement this in C, probably as an extmod). If I was going for unix only, then I could probably use FFI and just call open/close/ioctl.

So I thought I post here before starting and see what anybody else thinks. Initially, we only need to implement a fairly minimal open/close/read/write API. The biggest thing is that the open needs to put the serial port in raw mode. I'll look at the pyserial source to see exactly what it does,

pfalcon, 2015-05-24:
Serial support was implemented long ago: https://github.com/pfalcon/micropython-serial . A lot of other things were implemented long ago in https://github.com/micropython/micropython-lib . If someone needs to run it on a not POSIX compatible system, *then* they should come up with a proposal to do that (and the obvious way to add POSIX layer for their system).

stinos, 2015-05-24:
I think one can build FFI with mingw and use it to load libc so microptyhon-lib should be usable like that on windows, same for mac probably? If not or you want to support msvc then implementing it in C is likely the only option.

pfalcon, 2015-07-31:
@dhylands : Are you following/intend to follow this ticket? Otherwise, I guess it can be closed, and tickets for specific features can be submitted separately once hit.

dhylands, 2015-07-31:
@pfalcon I am, but I'm also happy to close this for now and reopen stuff later. It as more for discussion than anything else.

closed 2015-07-31 by dhylands