micropython/micropython #1284: esp8266:pyb.delay error

chinadsf, 2015-05-24:
when pyb.delay in range cycle by REPL, esp8266 will reboot.
[image]

pfalcon, 2015-05-26:
This is "expected", ESP8266 has quite different execution model from the normal model of Python. You're running in cooperative multitasking environment, so you cannot hog the control for a long time (like making a delay).

dpgeorge, 2015-05-26:
We could use the ESP RTOS to overcome this -- and all other -- problems that come from the coop multitasking, no?  Of course, someone would need to volunteer to do this :)

danicampora, 2015-05-26:
I think going for the RTOS is a good idea, it's the way is done with the CC3200. What was the reason for not doing it in the first place?

dpgeorge, 2015-05-26:
> What was the reason for not doing it in the first place?

Because there was no RTOS when I started the ESP port.

danicampora, 2015-05-26:
> Because there was no RTOS when I started the ESP port.

Ahh... that's a good reason ;-)

pfalcon, 2015-05-26:
> problems that come from the coop multitasking, no?

Coop multitasking is good, we just need to own the loop. Either set up our own, or "invert inversion of control" and own their loop. The latter looks particularly interesting, that's why I went to develop a stackless mode for uPy which is prerequisite for achieving that. But pragmatically/technically, setting up own loop is better, and I experimented into that too, in particular, I was able to run arbitrary pyb.delay() without adverse effects. That's where I stopped with esp8266 hacking, figuring I know enough of it to sleep at nights and wait till either licensing gets cleaned or a new exciting toy comes up ;-).

pfalcon, 2015-05-26:
> I think going for the RTOS is a good idea, it's the way is done with the CC3200. What was the reason for not doing it in the first place?

Because RTOS is bloat, and moreover come with strings attached. For example, do you know that you're prohibited to do any benchmarking on WiPy (well, publish them)? Because that's what FreeRTOS licence require: http://www.freertos.org/license.txt (Exceptions, Clause 2). Interesting that having such arbitrary restrictions on a use in a license makes FreeRTOS not an open-source software per OpenSource definition.
> I think going for the RTOS is a good idea, it's the way is done with the CC3200. What was the reason for not doing it in the first place?

Because an RTOS is bloat, and moreover comes with strings attached. For example, do you know that you're prohibited to do any benchmarking on WiPy (well, publish them)? Because that's what FreeRTOS licence require: http://www.freertos.org/license.txt (Exceptions, Clause 2). Interesting that having such arbitrary restrictions on a use in a license makes FreeRTOS not an open-source software per OpenSource definition. A GPL software which is not open-source is an ironic situation (akin to Espressif SDK).

danicampora, 2015-05-26:
> Because RTOS is bloat, and moreover come with strings attached.

Of course, nothing comes without a compromise, but it's a matter seeing if the pros outweigh the cons.

> For example, do you know that you're prohibited to do any benchmarking on WiPy (well, publish them)? 

No I didn't, but isn't the clause specific about relating FreeRTOS to the benchmark results? 
Funny that the most popular open source RTOS has such license restrictions...

pfalcon, 2015-05-26:
> but isn't the clause specific about relating FreeRTOS to the benchmark results?

But how you can separate what benchmark of your app are due to your app and what due to FreeRTOS? If you benchmark app builtin w/o FreeRTOS and the same app built with FreeRTOS, they have different performance, doesn't that imply something about FreeRTOS performance? Waste of resources required to answer such questions is the reason why nobody puts such clauses in open-source. But anyway, the conservative reading of the exception clause is just that - if it involves FreeRTOS, you can't benchmark it (well, until personally allowed).

> Funny that the most popular open source RTOS has such license restrictions...

I bet it's not only "the most popular", but also "the most performant", wink ;-).

owens-bill, 2015-06-13:
Since time.sleep() on the ESP8266 uses the same underlying functionality, it also causes crashes. Is there any mechanism that allows you to create a configurable delay?

owens-bill, 2015-06-21:
Can people still duplicate this problem? My recent firmware builds no longer seem to exhibit it, though I haven't figured out what changed.

sturmju, 2015-07-04:
Yes, the problem still occurs for me, both in the REPL mode as well as for the autorun script (scripts/main.py). 

Is there any workaround that I could use?

I would like to build a small IoT node that reads the temperature every 10seconds or so and publishes it to a webserver.

hamburger1984, 2015-12-11:
Running MicroPython v1.5.1-104-g4651c43 on ESP8266 here. The following snippet in REPL still causes a crash & reboot.

```python
>>> import time
>>> for i in range(10):
...     print(i)
...     time.sleep(1)
...
0
1
2
 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31036, room 16 
tail 12
chksum 0x9c
ho 0 tail 12 room 4
load 0x3ffe8000, len 924, room 12 
tail 0
chksum 0xed
load 0x3ffe839c, len 18932, room 8 
tail 12
chksum 0xb8
csum 0xb8
don't use rtc mem data
```

dpgeorge, 2015-12-12:
Yes, it's still not fixed....

pfalcon, 2015-12-12:
Fixing this (and other issues which preclude to use esp8266 port comfortably) requires almost complete rewrite of the port. It's known how to do that, but actually doing that is adifferent matter, and requires good deal of time/effort, and then testing, so keeps being in backlog.

dpgeorge, 2016-01-30:
This will be fixed when #1806 is closed.

dpgeorge, 2016-05-06:
Fixed by new execution model of ESP8266.

closed 2016-05-06 by dpgeorge