micropython/micropython #1285: Micropython not compiling on docker

sgoblin, 2015-05-24:
I was trying to compile micropython on alpine linux (alpine:edge) with docker, but it was failing with this message:

    /tmp # unzip -q v1.4.3.zip && cd micropython-1.4.3/unix && make && make test
    replace micropython-1.4.3/.gitattributes? [y]es, [n]o, [A]ll, [N]one, [r]ename: A
    Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
    mkdir -p build/genhdr
    CPP ../py/qstrdefs.h
    makeqstrdata ../py/qstrdefs.h qstrdefsport.h
    Traceback (most recent call last):
      File "../py/makeversionhdr.py", line 64, in <module>
        make_version_header(sys.argv[1])
      File "../py/makeversionhdr.py", line 13, in make_version_header
        git_tag = subprocess.check_output(["git", "describe", "--dirty", "--always"], universal_newlines=True).strip()
      File "/usr/lib/python2.7/subprocess.py", line 566, in check_output
        process = Popen(stdout=PIPE, *popenargs, **kwargs)
      File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
        errread, errwrite)
      File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
        raise child_exception
    OSError: [Errno 2] No such file or directory
    ../py/py.mk:132: recipe for target 'build/genhdr/mpversion.h' failed
    make: *** [build/genhdr/mpversion.h] Error 1

The following packages were installed: make, gcc, nano, unzip, readline-dev, libffi-dev, musl-dev, python-dev, pkgconf
All packages were up to date.

I later tried it on the ubuntu:vivid and ubuntu:trusty images, and it came with the same error:

    root@1634cf4b0f46:/tmp# unzip -q v1.4.3.zip && cd micropython-1.4.3/unix && make -j 4
    Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
    mkdir -p build/genhdr
    mkdir -p build
    mkdir -p build/py
    mkdir -p build/py/../extmod
    CPP ../py/qstrdefs.h
    Traceback (most recent call last):
      File "../py/makeversionhdr.py", line 64, in <module>
        make_version_header(sys.argv[1])
      File "../py/makeversionhdr.py", line 13, in make_version_header
        git_tag = subprocess.check_output(["git", "describe", "--dirty", "--always"], universal_newlines=True).strip()
      File "/usr/lib/python2.7/subprocess.py", line 566, in check_output
        process = Popen(stdout=PIPE, *popenargs, **kwargs)
      File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
        errread, errwrite)
      File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
        raise child_exception
    OSError: [Errno 2] No such file or directory
    make: *** [build/genhdr/mpversion.h] Error 1
    make: *** Waiting for unfinished jobs....
    makeqstrdata ../py/qstrdefs.h qstrdefsport.h

The following packages were installed: libffi-dev, build-essential, unzip, wget, libreadline-dev, pkg-config
All packages were up to date.

When I tried to compile it on an Ubuntu Trusty VM on DigitalOcean, it worked fine (all tests passed).

Why is it not working on Docker?

dpgeorge, 2015-05-24:
It's because git is not installed and py/makeversionheader.py needs git to work out the version number.

We need to find a way around this.... but I don't know how to elegantly find the version number without using git.  Maybe parse docs/conf.py?

pfalcon, 2015-05-24:
With a new title, bug is not fixed.

sgoblin, 2015-05-24:
I would suggest having a version file in the downloads for full version numbers, and check for that. For development versions, maybe just using git, since you will probably have that installed while developing anyways.

dpgeorge, 2015-05-25:
Fixed.  @sgoblin are you able to check the latest master version compiles for you?

sgoblin, 2015-05-25:
It compiled from the master tarball on an ubuntu:vivid docker container, all tests passed, one skipped:

    433 tests performed (15787 individual testcases)
    433 tests passed
    1 tests skipped: extra_coverage

Ran: apt update && apt upgrade && apt install python python3 libffi-dev build-essential unzip wget libreadline-dev pkg-config

closed 2015-05-25 by sgoblin