micropython/micropython #1293: esp8266: can't read from sys.stdin

deshipu, 2015-05-28:
I was trying to work around the fact that the ESP8266 port doesn't have UART support on the Python side yet, and simply read/write from and to the sys.stdin and sys.stdout.

While sys.stdout works as expected, trying to read even a single character from sys.stdin hangs the REPL. I compared with other ports, and that's not the case in them -- they wait until the specified number of characters is input and then properly return those characters.

deshipu, 2015-05-28:
Also doesn't work from main.py -- the board just hangs.

danicampora, 2015-05-28:
Is the stdio uart in the esp8266 port interrupt driven?

dpgeorge, 2015-05-30:
With the changes that make the REPL event driven, the function `mp_hal_stdin_rx_chr` no longer works at all (because received chars are given directly to the REPL and the uart rx buffer never gets filled).

dpgeorge, 2016-05-06:
This has been fixed with the new ESP8266 execution model.

closed 2016-05-06 by dpgeorge