micropython/micropython #1294: Repeated calling of map() crashes pyboard

peterhinch, 2015-05-29:
The following function runs correctly if called with small values of its argument. Large values (1000) cause a crash. Sometimes it fails more gracefully with a memory allocation failure. The cause appears to be the map() line.

```python
count = [0, -10, 10]

def foo(n):
    global count
    magmax = count[:]
    for x in range(n):
        magxyz = tuple(count[:])
        magmax = map(max, magmax, magxyz)
        count = [x +1 for x in count]
    return tuple(magmax)
```
Tested by running bar.foo(1000) at the REPL.

danicampora, 2015-05-29:
I hit a hard fault on the CC3200. Looks like stack overflow.

Anton-2, 2015-05-29:
This can crash CPython 3.4 too, given a large enough n.

In python 3, and micropython, map is an iterator object. Here each map object depends on the previous map object, so they all stay allocated until used (`return tuple(magmax)`).

You can avoid this by 'materializing' the results :
```python
magmax = list(map(max, magmax, magxyz))
```

or using list comprehension.

pfalcon, 2015-05-29:
> This can crash CPython 3.4 too, given a large enough n.

Is there bugreport for that?

And we need to fix the crash and throw exception instead. If you know already where stack check fix is missing, please propose a patch. Otherwise, next step is reproducing on unix port.

dpgeorge, 2015-05-29:
Interesting stuff, nice find!

peterhinch, 2015-05-30:
@Anton-2 Thanks for that: interesting implications of map() returning an iterator. This could affect code ported from Python 2 to P3. My original code has an iteration count which varies at runtime. Such a function would pass tests for return value validity, later throwing a runtime error when the iteration count exceeded an arbitrary limit. Nasty bear trap.

dpgeorge, 2015-06-03:
The crash is indeed due to stack overflow.  It's because map_iternext calls mp_iternext which calls type->iternext which dispatches to the next nested map_iternext, and so on.  enumerate, filter and zip are also affected in the same way.

Fixed in 953c23b1bcc732ccb3a6e4798add9b1f2fc0f444.

With the patch pyboard can do foo(152) at most, foo(153) will lead to a RuntimeError exception for stack overflow.

unix port can do around 500 iterations.

CPython does not seem to crash, tested up to 10000 iterations (it can reallocate the stack).

closed 2015-06-03 by dpgeorge