micropython/micropython #1295: Generate standalone binaries of uPy?

dpgeorge, 2015-05-29:
There has been some talk recently on python-dev about making standalone binaries of CPython, specifically to run on Windows.  This is something that uPy can do very easily and it would be nice and compact.

We could generate (automatically and daily) standalone binaries for Windows, OSX and Linux which are a single executable file and "just run" without any installation or dependencies.

Is this interesting?

pfalcon, 2015-05-29:
Did the guys wanted *that*? I figure, they wanted single-executables of complete *apps*. As my reply there showed, that's interesting, and something I have in my todo list for a long time.

dpgeorge, 2015-05-29:
> Did the guys wanted that?

Not that in particular, but it's a simple and natural precursor.  Eg maybe you have a fresh system (Windows, Linux whatever) without any nice tools and you need to do some quick hacking.  Being able to get a standalone Python executable might be useful.

> ... wanted single-executables of complete apps. As my reply there showed, that's interesting, and something I have in my todo list for a long time.

We should have a separate ticket for that.  But, as you say, there are road blocks first to overcome and they have their own issues.

pfalcon, 2015-05-29:
> Being able to get a standalone Python executable might be useful.

Python is not "executable", Python is "batteries included". Even if someone asked for what you say, they asked for Python and standard library packages into single executable file. That's different from just "Python executable". We simply don't have a complete standard library. And I dedicated May exactly to having a progress with micropython-lib, even though "progress" largely consists of charting the space - I add any useful module, even if module's implementation is empty so far (but as I reported on forum, I actually was able 2 third-party libs with minimal changes now). So, someone please come help me with micropython-lib work.

dpgeorge, 2015-05-29:
> Even if someone asked for what you say, they asked for Python and standard library packages into single executable file.

We have a small (useful) selection of included libraries built in (eg sys, ure, etc).  I use uPy daily on Linux without the need for anything from micropython-lib (thanks mainly to modffi).

pfalcon, 2015-05-29:
> I use uPy daily on Linux without the need for anything from micropython-lib (thanks mainly to modffi).

Let me say that's bad and this way we will never conquer the world and stay just a niceh Python-like language, of ~ zero interest to general Python community. We need push ourselves to be really Python workalike, not just something which has Python syntax.

Anyway, to get to productive discussion for this ticket, what we need to do then is to put upip into such executable as a frozen module. Then yes, it won't be just a funny demo, but will be actually useful. More details: https://mail.python.org/pipermail/python-dev/2015-May/140351.html

We also generally need to PR better. For example, what about putting mentioning of micropython-lib/upip in micropython's README?

pfalcon, 2015-05-29:
Of related notice: @roger- , how micropython/micropython-lib package maintance for OpenWRT goes?

roger-, 2015-05-29:
@dpgeorge 

I like the idea of standalone binaries. Would be great if a buildbot could make them for every platform, if that's feasible.

@pfalcon 

It's still on the latest tagged version (v0.1). 

MicroPython is running behind, so I'll have to do some testing and update it soon.

dpgeorge, 2015-05-29:
> Let me say that's bad

Why?  It shows that uPy is already useful and can do real work without need for a lot of libraries.  I use modffi to hook into external C++ code that I've written to compute cpu-intensive physics stuff, and uPy does the high level stuff.

> We need push ourselves to be really Python workalike, not just something which has Python syntax.

uPy has both syntax and *semantics* (which is already non trivial).   The next step is the library, both micropython-lib and defining and writing hardware peripheral drivers for microcontrollers.

+1 for work on micropython-lib, but we are already stretching ourselves with work on a solid core, work on pyboard, wipy etc, and work on unix, Windows and OSX ports.  There's only so much a small team can do.  The main aim is to run well on microcontrollers -- no one else can do that.

dpgeorge, 2015-05-29:
> For example, what about putting mentioning of micropython-lib/upip in micropython's README?

upip needs good support for pyboard (and wipy).  Last time I tried the library on pyboard it had all sorts of problems because it overrode sys and os with versions that tried to use ffi.

danicampora, 2015-05-29:
> The main aim is to run well on microcontrollers -- no one else can do that.

+1.

pfalcon, 2015-05-29:
> It's still on the latest tagged version (v0.1).

Hmm, I thought we discussed date releases, like 0.1-20150501. Anyway, I intend to tag 0.5 soon, please subscribe to https://github.com/micropython/micropython-lib/issues/23 to be in loop.

> MicroPython is running behind, so I'll have to do some testing and update it soon.

Thanks for keeping up this work.

pfalcon, 2015-05-29:
>>  Let me say that's bad

> Why? It shows that uPy is already useful and can do real work without need for a lot of libraries. I use modffi to hook into external C++ code that I've written to compute cpu-intensive physics stuff, and uPy does the high level stuff.

Ok, if you plug into your custom C/C++ code, that's good. But most people don't use Python like, one usecase is writing small, often adhoc scripts. But even that requires standard library modules, if not 3rd-party stuff.

> +1 for work on micropython-lib, but we are already stretching ourselves with work on a solid core, work on pyboard, wipy etc, and work on unix, Windows and OSX ports. There's only so much a small team can do. The main aim is to run well on microcontrollers -- no one else can do that.

And yet this ticket. So what I argue, is that if doing that, that doing it so it was actually useful for outside people, not just kind of demo we do because we can. Doing that in useful manner requires some effort, yeah. One way to approach that is to have kind of "sprints", like we say "let's work next month to progress area X", where X is not "core" baremetal "place". E.g. achieving self-hostedness of micropython own tools (#1282) sounds like good idea for such sprint.

pfalcon, 2015-05-29:
> upip needs good support for pyboard (and wipy). Last time I tried the library on pyboard it had all sorts of problems because it overrode sys and os with versions that tried to use ffi.

Ok, @dpgeorge , @danicampora go for it on your boards ;-).

Seriously, no, upip isn't going to work on boards until SSL problem is solved. And no need to put all that on critical path. Again, it already works on unix, let's make its usage seamless and comfortable. When it will be actually used, we can look how to progress it further and extend its coverage.

Summing up, above I asked specific questions about next steps re: pip, I'd appreciate the answers (all that is already on my todo list, it's just get constantly backlogged, because I don't have a feeling that it interests somebody else but me).

pfalcon, 2015-05-30:
> Anyway, to get to productive discussion for this ticket, what we need to do then is to put upip into such executable as a frozen module.

I hope the answer is affirmative, because patch is already prepared: https://github.com/micropython/micropython/pull/1296

dpgeorge, 2015-05-30:
> Anyway, to get to productive discussion for this ticket, what we need to do then is to put upip into such executable as a frozen module.

It's not obvious to me that this is the right approach.  Let's move the discussion of this point to #1296.

dpgeorge, 2015-05-30:
> We also generally need to PR better. For example, what about putting mentioning of micropython-lib/upip in micropython's README?

First we should mention micropython-lib itself, since that's what is really interesting to people.

dpgeorge, 2015-05-30:
> One way to approach that is to have kind of "sprints",

Yes, good idea to have some focus rather than knee-jerk reacting to bugs/features/PRs.  But organising such things is the difficult bit.

dpgeorge, 2015-05-30:
> Seriously, no, upip isn't going to work on boards until SSL problem is solved.

Actually I meant that upip doesn't work *for* the pyboard (ie running on unix installing onto pyboard) because it doesn't know that the pyboard needs different versions of the libraries; eg micropython-os doesn't work on pyboard and micropython-errno uses the wrong numbers.  We will need different versions of these libraries and a way for upip to know what platform it's installing for.

pfalcon, 2015-06-22:
Ok, so with upip builtin, and even documented somewhat, the original idea of this ticket looks more attractive, I guess it can be scheduled per @dpgeorge's availability to do it ;-).

But then there're more ideas too. E,g,, after getting rid of libreadline dependency, the only external dependency for unix is libffi. We could follow our approach to pull in dependencies and link against it statically (or link statically without pulling in, though I'd follow the most general approach).

dpgeorge, 2015-06-22:
Linking uPy statically (with -static) gives an error becaule dlopen can't be made completely static (apparently).  Also getaddrinfo has similar problems.

pfalcon, 2015-06-22:
Yep, stupid glibc, would need to give up it too (which would need to be done anyway, because seeing glibc static sizes is not for the faint of heart).

dpgeorge, 2015-06-22:
I think there's still some way to go before giving up glibc for our own versions of all libc required functions.

We need ffi otherwise upip won't work.  So the best we can do right now is generate binaries with shared libraries, which means they'll run only on certain machines.

pfalcon, 2015-06-24:
> I think there's still some way to go before giving up glibc for our own versions of all libc required functions.

Just to clarify, I don't mean to having our own complete libc - that's not possible and not practical, just switching to another libc for such builds, less bloated than glibc. And that's certainly not on critical path to just starting to make such builds, just a logical extension of the idea ;-).

pfalcon, 2015-08-30:
Step towards producing truly standalone binaries is done in https://github.com/micropython/micropython/commit/a9058bf2944db20f5e2377a6a38c37a46d3d6aa0 , which allows to build libffi for any target and link with it statically.

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge