micropython/micropython #1297: Move all loose methods inside modpyb to a single class.

danicampora, 2015-05-31:
We should move reset, unique_id, interrupt, sleep and so on, to a single class (`machine` for instance) to handle all "microcontroller" and non-standard python things. That way modpyb ends up being a clean collection of classes. We should also keep in mind to not make `machine` the place to put all those things that do not fit anywhere else ;-).

danicampora, 2015-05-31:
Some of the methods that I don't see being part of `machine` are the time related ones (elapsed_mililis, elapsed_micros) any chance we can move those to modutime? I know is not standard stuff, but we can make extensions as long as they make sense, right?

pfalcon, 2015-05-31:
There's nothing particularly "clean" in putting everything to classes. Python is not Java (thanks god), so it makes little sense to enforce artificial restrictions like that. However, there's a "machine" *module* now, and all those functions really should go there (I skipped mentioning this on @dpgeorge's related comment yesterday, but if you raise it as an additional ticket, here it is).

pfalcon, 2015-05-31:
Please note that "machine" module is intended to by "pyb" v2-final, so anything going there should be given real good thought.

pfalcon, 2015-05-31:
> Some of the methods that I don't see being part of machine are the time related ones (elapsed_mililis, elapsed_micros) any chance we can move those to modutime? I know is not standard stuff, but we can make extensions as long as they make sense, right?

Yes, moving those to "time"/"utime" is what I had in mind to propose too. The closest ticket to discuss that is https://github.com/micropython/micropython/issues/1225 (because again, we should apply good thought to add to "time" good and consistent API, I even have an idea to run it thru python-ideas list).

dpgeorge, 2015-05-31:
> There's nothing particularly "clean" in putting everything to classes.

I think @danicampora did in fact mean module.

It must be said though: I do like the pyb module being a one-stop-shop for all things microcontroller related.  Splitting the functions/classes out into different modules means people need to remember both the module name and the function/class name, instead of just knowing to always use pyb.

Having functionality split over many modules also requires good documentation.  If time-related functions go in utime then we need to highlight the differences between standard time and utime, pointing out the special functions that are more suited to a microcontroller (eg utime.delay, utime.elapsed_millis).  At the moment it's easy to just say "all special functions are in pyb".

danicampora, 2015-05-31:
> There's nothing particularly "clean" in putting everything to classes. Python is not Java (thanks god), so it makes little sense to enforce artificial restrictions like that.

Well, it's clean for me ;-). Anyway the idea was not to put restrictions but to organise things better.

danicampora, 2015-05-31:
> Having functionality split over many modules also requires good documentation

Well, we are already discussed about making improves in that area, and things are being done.

roger-, 2015-05-31:
>  I do like the pyb module being a one-stop-shop for all things microcontroller related

Since a lot of the functions are common to other platforms, shouldn't the `pyb` module be renamed to something more general?

pfalcon, 2015-05-31:
> It must be said though: I do like the pyb module being a one-stop-shop for all things microcontroller related.

Is it pythonic to call it like that? Did you run that idea thru (e.g.) python-ideas, what feedback from teh general Python community was?

> Splitting the functions/classes out into different modules means people need to remember both the module name and the function/class name, instead of just knowing to always use pyb.

Poor Guido guy, he could have done much better by putting all Python functionality in one module named e.g. "foobar" ;-).

>  If time-related functions go in utime then we need to highlight the differences between standard time and utime, pointing out the special functions that are more suited to a microcontroller (eg utime.delay, utime.elapsed_millis). 

Yes, and https://github.com/micropython/micropython/issues/983 is one of the oldest docs-related issue.

> At the moment it's easy to just say "all special functions are in pyb".

pyb.py:
~~~~
from * import *
~~~~

danicampora, 2015-06-07:
Well, what to do then? 

1. Put IRQ, Sleep and reset handling inside modmachine?
2. Do we want to get rid of modpyb completely? and then ADC, SPI, I2C, etc. would all be independent modules?

dhylands, 2015-06-07:
I wouldn't want to get rid of pyb until such time as I can use dir on some global dictionary to discover what's there. Presumably such a dict exists, and we just need to expose it as sys.modules?

pfalcon, 2016-06-19:
Superseded by https://github.com/micropython/micropython/issues/1430.

closed 2016-06-19 by pfalcon