micropython/micropython #1300: uctypes.struct() argument order is inconsistent and error-prone

pfalcon, 2015-06-02:
Currently, there's following signatute:

uctypes.struct(descriptor_dict, addr, layout_type=NATIVE)

addr and layout_type are both ints.

So, "addr" is stuck in the middle of type-definining parameters. This of course because layout_type has a default value. Moreover, both addr and layout_type are ints, which leads to not reporting errors for clearly incorrect usage: http://forum.micropython.org/viewtopic.php?f=2&t=723

So, there're 2 ways to solve it:

1. Move "addr" param to be first. This also fixes grouping of semantically different params.
2. Require 3 args all the time. That makes it verbose, though perhaps more explicit.

Comments welcome.

dpgeorge, 2015-06-02:
I'd opt for option 1.  The address is arguably the most important thing, and you can then read it as "take this addr and interpret it using this type".

Note that you could use strings to specify the layout_type: ` uctypes.struct(0x1234, {...}, "NATIVE")`.  The qstr for NATIVE already exists so having it in the script doesn't increase RAM usage.

pfalcon, 2015-06-02:
> I'd opt for option 1. The address is arguably the most important thing, and you can then read it as "take this addr and interpret it using this type".

I also favor 1, even though it's not backwards-compatible. The original idea though was that type is the most important thing ;-). I'll enqueue the refactor.

> Note that you could use strings to specify the layout_type: uctypes.struct(0x1234, {...}, "NATIVE").

My general consideration of such things is that they are luaish/javascriptish hacks. Having interned strings is clearly an implementation detail in Python. But perhaps I should look to find positive things in it (like automagic human-readability on output), and then think about it as a whole picture.

dpgeorge, 2015-06-02:
> I also favor 1, even though it's not backwards-compatible.

Option 2 also requires code modification (in general).

> But perhaps I should look to find positive things in it ...

Strings are used in some places (eg "little" for little endian in int (un)packing) and file opening mode.

Benefits of strings:
- no proliferation of constants in a class (which can slow method lookup); saves ROM
- smaller parse tree (class.CONST takes a node and 2 leaves, while "CONST" is just a leaf)
- smaller and faster bytecode (unless constants are optimised in the compiler, not the case for uctypes)

Benefits of constants:
- can "or" them together if needed
- general no conversion needed to go to the value used internally to select the feature (eg Pin.PULL_UP)
- tab completion (in most cases)

Both have:
- good readability and self documenting
- value checking (strings in the function being called, constants when doing load attr on the base)

pfalcon, 2015-06-02:
> can "or" them together if needed

Right, that's exactly what I was thinking of yesterday - that we'd need NATIVE | AUTOSIZE to support automatic offset calculation for fields (as would be enabled by OrderedDict).

pfalcon, 2015-06-06:
Fixed. But the first thing this change broke was upip (should be fixed now too). And likely not the last...

closed 2015-06-06 by pfalcon