micropython/micropython #1301: MP_QSTR_sd' undeclared when 'make BTARGET=application BTYPE=release BOARD=LAUNCHXL'

garyatpt, 2015-06-02:
xxx@xxx:~/micropython/cc3200$ make BTARGET=application BTYPE=release BOARD=LAUNCHXL

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/LAUNCHXL/release/genhdr
.
.
CC mods/modpyb.c
CC mods/moduos.c
In file included from mods/moduos.c:33:0:
mods/moduos.c: In function 'os_listdir':
mods/moduos.c:154:58: error: 'MP_QSTR_sd' undeclared (first use in this function)
             mp_obj_list_append(dir_list, MP_OBJ_NEW_QSTR(MP_QSTR_sd));
                                                          ^
../py/obj.h:84:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^
mods/moduos.c:154:58: note: each undeclared identifier is reported only once for each function it appears in
             mp_obj_list_append(dir_list, MP_OBJ_NEW_QSTR(MP_QSTR_sd));
                                                          ^
../py/obj.h:84:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^
make: *** [build/LAUNCHXL/release/mods/moduos.o] Error 1

danicampora, 2015-06-02:
SD card support is disabled for the LAUNCHXL since there's no SD card in that board. You can enable it by defining `MICROPY_HW_HAS_SDCARD` to one in `cc3200/boards/LAUNCHXL/mpconfigboard.h`

dpgeorge, 2015-06-02:
But this still needs fixing by wrapping lines 153-155 of moduos.c in `#if MICROPY_HW_HAS_SDCARD`.

danicampora, 2015-06-02:
True, just saw it, I'll commit in a minute.

danicampora, 2015-06-02:
Fixed in https://github.com/micropython/micropython/commit/2ffb6e1b157b856178426c1d040a9181174c5976

closed 2015-06-02 by danicampora