micropython/micropython #1302: readthedocs lacks 1.4.3 docs

pfalcon, 2015-06-02:
See https://readthedocs.org/builds/micropython/ , 1.4.2 is last there.

pfalcon, 2015-06-02:
Ok, it lacked it, because nobody built it in a month. I triggered "latest" build, and it picked up 1.4.3 tag.

closed 2015-06-02 by pfalcon

dpgeorge, 2015-06-02:
Yeah, it's not the most reliable thing at auto-updating.  This issue will go away soon when we move to self-hosted docs.

pfalcon, 2015-06-02:
> This issue will go away soon when we move to self-hosted docs.

If timely updates can be guaranteed (and delay figures for it disclosed). Otherwise, having a publicly hosted/controlled hosting clearly helps with cases like http://forum.micropython.org/viewtopic.php?f=2&t=723 - I can make a quick fix and "deliver" it.

dpgeorge, 2015-06-02:
> If timely updates can be guaranteed

It can be done each time a new commit is pushed.

hickford, 2017-02-10:
I noticed there are two copies of the documentation:

1. http://docs.micropython.org/en/latest/pyboard/
2. https://micropython.readthedocs.io/en/latest/

1 looks to be automatically updated, whereas 2 is a few weeks behind. 2 has handy 'Edit on GitHub' links which are missing from 1.

dpgeorge, 2017-02-10:
The readthedocs copy of the documentation is no longer supported (even though it does update from time to time).  If you like, feel free to make a contribution adding the "Edit on GitHub" link to the docs in this repo, then it'll be available at docs.micropython.org.