micropython/micropython #1304: cc3200: Connection closed by foreign host.

garyatpt, 2015-06-03:
My board is v4.1 LAUNCHXL board.
Flashing the latest service pack (servicepack_1.0.0.10.0.bin).
I have built both the bootloader and the application in release mode:
 $make BTARGET=application BTYPE=release BOARD=LAUNCHXL
 $make BTARGET=bootloader BTYPE=release BOARD=LAUNCHXL
I access the MicroPython REPL through telnet.

I wrote a simple test program:
i=0
while 1:
....i+=1
....print(i)

After performing for some time every time,I see the following: 
Connection closed by foreign host.

Why would I get this error?

================================================
xxx@xxx:~$ telnet 192.168.1.1
Trying 192.168.1.1...
Connected to 192.168.1.1.
Escape character is '^]'.
Micro Python v1.4.3-102-ga546acd on 2015-06-03; LaunchPad with CC3200
Login as:micro
Password:
Login succeeded!
Type "help()" for more information.

>>> i=0
>>> while 1:
...     i+=1
...     print(i)
...
1
2
3
4
.
.
.
163096
163097
163098Connection closed by foreign host.
xxx@xxx:~$

danicampora, 2015-06-03:
I think I know the reason, and I also think I already have a proper solution for it. I'll come back with more info later today. Thanks!

danicampora, 2015-06-03:
There's a 5 minute timeout on the telnet server and 2 minutes on the ftp server. After that time of inactivity the connection is closed. This is done to avoid half-open connections, which occur for example when the WiFi connection is lost without notification. I have a fix for this which implies resetting the servers when I receive a WiFi connected event. This solves the half open connection issue in the scenario when the WiFi connection is suddenly dropped, but what if the telnet client on the PC hangs, and it's closed without properly closing the socket? I guess we have a problem again.... So, there are two options:

1. Remove the timeout completely.
2. Increase the timeout to lets say... 10min, 30min?

Thoughts?

pfalcon, 2015-06-03:
The way it's done "professionally" is to have keep-alive support. To work reliably, this should be part of application protocol (because only particular application knows what's good timeout for it). Failing that, there're transport-level (TCP) keep-alives, but those are set conservatively high (hours, to not disturb unsuspecting applications). Quick googling doesn't show keep-alive support for telnet, like e.g. ssh has ;-(. I suggest to do more research to see if it's possible (and in server->client direction, as needed here), and if not, make it configurable and rely on TCP keep-alive in worst-case (possibly see if those can can be made shorter).

danicampora, 2015-06-03:
> Quick googling doesn't show keep-alive support for telnet, like e.g. ssh has ;-(. I suggest to do more research to see if it's possible (and in server->client direction, as needed here), and if not, [...]

I did more than a quick googling and I found the same. Keep alive on TCP side is really high and  not easy to configure. Then the keep alive support on application level is not standard for telnet clients. FTP clients do have an option for this (list the directory every 30 secs for example), but not all of them.

In my opinion we should have both the timeout and the servers reset mechanism on the WiFi connected event. I could add a method to make the timeout configurable and leave 5 min as default. What do you think? 

The only "professional" solution for this is to have our own telnet and ftp clients that support a proper keep alive mechanism. But we have no resources for this right now and I don't want to force people to use a specific tool.

EmmBee, 2015-06-03:
In days of old, when men were bold and telenet was not invented, IBM would send the ASCII ENQ charater to conneted "tty's" only IBM supplied (& modified) "tty's" knew to answer with 'ACK' to keep the Teletype channel open.
How do I know ? Guess..

danicampora, 2015-06-03:
> In days of old, when men were bold and telenet was not invented, IBM would send the ASCII ENQ charater to conneted "tty's" only IBM supplied (& modified) "tty's" knew to answer with 'ACK' to keep the Teletype channel open.
How do I know ? Guess..

Yes that's an option, but then we can't trust every telnet client to work like this, we would need a custom made client. That's the main problem.

pfalcon, 2015-06-03:
> Yes that's an option, but then we can't trust every telnet client to work like this, we would need a custom made client. That's the main problem.

Not as I see it. What we need to do is to send from us (server) to client: 1) either a command which will be ignored, or 2) a "character" which won't be displayed to user. We don't really even need application-level ack for that - ack is required on TCP level, and non-receiving it will be detected in much shorter time.

danicampora, 2015-06-03:
> 1) either a command which will be ignored, or 2) a "character" which won't be displayed to user. 

OK, sounds good. I will try that approach and see how reliable it is.

danicampora, 2015-06-03:
> 1) either a command which will be ignored, or 2) a "character" which won't be displayed to user.

But, I remember I already tried this and with a half open connection sending succeeds (at least it thinks it succeeds). I can test it again this evening, but my hopes are low...

EmmBee, 2015-06-03:
quite !
And like IBM you quietly close the port - forever (or until OS reboot) ;-}
You just can't win - ever 'cos there are ASCII enq's and ack's in each and every picture file.....
I was just pointing out the problem was not new - it was extant in 1978.

EmmBee, 2015-06-03:
If you use something like enq/ack, deep down (almost below) the transport level, remember to escape and unesape real "enq" & "ack" characters in the client data stream.

danicampora, 2015-06-03:
With https://github.com/micropython/micropython/commit/cc20482aa997af7c20fe4ec9cad85762aa52b458 I attempt to improve the robustness of the server against half-open connections, according to my tests it does. Anyway there's a default timeout of 300 secs which can now be changed by `network.server_timeout(secs)`. This method can also be used to query the current timeout configured.

@garyatpt please pull the latest changes and let me know if this works for you.

garyatpt, 2015-06-03:
Thanks！I'll test it later today.

garyatpt, 2015-06-04:
Okay,the range of network.server_timeout(secs) can be set from 5 seconds to 4294967 seconds(about 49.71 days).
It works for me currently.
Thanks!

danicampora, 2015-06-04:
Cool!, thanks for the feedback :-)

closed 2015-06-04 by danicampora