micropython/micropython #1306: unix: difference to CPython when catting script into stdin

dpgeorge, 2015-06-03:
In CPython you can cat the script into stdin:
```
cat 'print(123)\nprint(456)' | python
```
This works as expected.  In uPy if you do this then it enters the REPL and the stdin data is interpreted at the REPL.

I think we should follow CPython here.  It's more unixy.

pfalcon, 2015-06-03:
Sounds good, isatty(), etc.

dpgeorge, 2015-06-03:
It'll break all tests in tests/cmdline... but I'm prepared to deal with that :)

dpgeorge, 2015-06-03:
Ok, it's pretty trivial to implement this using the following:
```c
if (!isatty(0)) {
    do_file("/dev/stdin");
}
```
But that probably won't work on windows... does windows have isatty?  I'm pretty sure it doesn't have /dev/stdin :)

Other option that will work on windows is to make a new wrapper for the lexer to init a file from a file descriptor.  But that adds bulk.  @stinos any suggestions?

stinos, 2015-06-04:
Windows has isatty() and it works fine as far as I can tell.
I see no other way to read from stdin then to, well, just read from stdin - through it's descriptor that is.
So if you don't want to add a ```lexer_from_fd``` or so, I don't have any other suggestion than to make ```open()``` for the win port to return the stdin descriptor when passed the /dev/stdin string instead of trying to actually open from the filesystem.

dhylands, 2015-06-04:
I would be inclined to make lexer_from_fd and have it also take a filename argument for error reporting purposes. Then you can use stdin or pipes or whatever.

Make the existing lexer function open the file and pass the filename in. Hardly adds any extra code.

dpgeorge, 2015-06-04:
I agree with @dhylands to provide a way to open directly a file descriptor.

dpgeorge, 2015-06-04:
Implemented.

closed 2015-06-04 by dpgeorge