micropython/micropython #1307: cc3200: Watchdog can't cause timeout

garyatpt, 2015-06-04:
I enable the watchdog timer with 5000msec timeout value:
>>>my_wdt=pyb.WDT(5000)

For the case one,watchdog can cause timeout and reset the board without kicks the watchdog timer.
But for the case two,the watchdog can't cause timeout. Why?

=============================================================================
CASE 1: 
>>> import pyb
>>> my_wdt=pyb.WDT(5000)
>>> my_wdt.kick()
>>> i=0
>>> start=pyb.millis()
>>> while 1:
...     if (pyb.elapsed_millis(start))>=1000:
...         start=pyb.millis()
...         i+=1
...         print(i,'sec')
... 
1 sec
2 sec                                                                           
3 sec                                                                           
4 sec                                                                           
5 sec
^^^
Watchdog cause timeout and reset the board.

===========================================================================
CASE 2:
>>> import pyb                                                                  
>>> my_wdt=pyb.WDT(5000)                                                        
>>> i=0                                                                         
>>> while 1:                                                                      
...     i+=1                                                                    
...     print(i)                                                                
... 
1                                                                         
2                                                                           
3                                                                           
.
.
.
1313844                                                                         
1313845                                                                         
1313846                                                                         
1313847 
.
.
^^^
Keep counting when over 5 seconds.......Why?

danicampora, 2015-06-04:
Interesting... I'll investigate...

danicampora, 2015-06-04:
Thanks for the report. I tried to make WDT usage simpler and made it work in a rather unusual way in the sense that it would be kicked in the background if things were perceived as working "properly", but I realize that this is not correct and there are more implications. So, once enabled, the WDT must be kicked by the user, otherwise it will reset.

Fixed in https://github.com/micropython/micropython/commit/a3cf4ea2f69e3568944bad7ee1a017b7095efc3d

closed 2015-06-04 by danicampora