micropython/micropython #1308: cc3200: The cc3200 chip crashes ﻿when I use the finger touch crystal Y3 on LAUNCHXL-board

garyatpt, 2015-06-05:
When I use the finger touch crystal Y3 on LAUNCHXL-board, the cc3200 crashes and the computer REPL screen freezes, the heartbeat-light no longer flash.
Even though I have enabled WDT, the situation is still the same.
If WDT have effect, the board should be reset. Right?

===============================================================================
TEST CODE:

import pyb                                                                  
my_wdt=pyb.WDT(5000)                                                        
i=0                                                                         
while 1:                                                                      
....my_wdt.kick()
....i+=1                                                                    
....print(i)                                                                
.... 
1
2
3
.
.
131383
131384                                                                         
^^^
Touching the crystal Y3, cc3200 crashes and computer REPL screen freezes.

danicampora, 2015-06-05:
@garyatpt Thanks for the bug report. However, in my case the WDT is working properly. I can reproduce the behavior of the CC3200 crashing when touching the 40MHz XTAL with my finger, but (using your test code), the WDT resets the SoC and it boots-up again. 

One question... are you testing this while debugging with CCS and the SOP2 jumper in place? If that's the case after being reset by the WDT the CC3200 will enter UniFlash programming mode. You need to build release versions of the application and the bootloader, program them with UniFlash (as described in cc3200/README.md), remove SOP2, reset the board and then execute your test again. 

You cannot test this kind of things that involve a hard reset while in a debug session because the CC3200 executes all code from RAM.

garyatpt, 2015-06-05:
After programming the image of application and the bootloader, the SOP2 jumper is removed (SOP2,SOP0 too).
I did not do any debugging using TI-CCS while testing.
I use FTP download my test code and perform.
Then touch, crash....:(

danicampora, 2015-06-05:
Yes, it does crash but then it will reboot after ~5s. In my case it always reboots (well, at least the ~20 times I have tried your test so far..)

garyatpt, 2015-06-05:
Maybe I should show the video to you.
Waiting for me to upload.

danicampora, 2015-06-05:
Sure, I'll check your video.

danicampora, 2015-06-05:
@garyatpt did you update you build the latest code? Do you reset the board after updating via the FTP server?

garyatpt, 2015-06-05:
did you you build the latest code?
>YES

Do you reset the board after updating via the FTP server?
>YES

Please see the link and download file (57.7MB):
https://drive.google.com/file/d/0Bx3vAAbz03kBTVJGdjQxYXphZWM/view?usp=sharing

danicampora, 2015-06-05:
Really weird. I will upload a video a bit later with my results... I'll check with TI, let's see what they say...

danicampora, 2015-06-05:
@ garyatpt please check my video:

https://drive.google.com/file/d/0B8OL5q881yLGU2VzVEVBS1NMcHM/view?usp=sharing

garyatpt, 2015-06-06:
Is it possible to enter an infinite loop when the cc3200 receives some fault
interrupt ? (and watchdog also died)

Can we get some debugging information from 'micropython/cc3200/hal/startup_gcc.c' ?

danicampora, 2015-06-07:
You can add some additional code to `HardFault_HandlerC()` in `startup_gcc.c` like blinking some of the launchxl LEDs. But still, the watchdog should cause a reset if the code ends up there.

I really don't know how to continue here, I have 2 LAUNCHXL boards, both behave correctly. Also tested on several WiPy's, all of them work as expected.

dpgeorge, 2016-12-02:
Closing because this issue is very old, and wasn't properly reproducible.

closed 2016-12-02 by dpgeorge