micropython/micropython #1309: Minifier for frozen modules - write own or use existing

pfalcon, 2015-06-06:
Before writing my own minifier I went to study prior art and e.g. found https://github.com/liftoff/pyminifier , which doesn't just compresses indents (trivial), removes whitespace (supposedly easy), but also renames identifiers (I'd probably skip writing that). W/o the latter step it gives figure of 1358 -> 640 bytes compression (also kind of info I was looking for).

dpgeorge, 2015-06-06:
I always err on the side of do it yourself/myself, but renaming identifiers is pretty nifty and non-trivial, so might be sensible to use that pyminifier.

Other option would be to compress frozen modules and decompress on the fly when lexing (pretty easy to do).  Then minification is not needed.

pfalcon, 2015-06-06:
> Other option would be to compress frozen modules and decompress on the fly when lexing (pretty easy to do).

Really? Depends on compression method I'd say. uzlib still doesn't have stream operation mode, even though it's "easy".

dpgeorge, 2015-06-06:
> uzlib still doesn't have stream operation mode,

Oh, right, well then it's not that easy.  Interfacing to the lexer with an arbitrary stream is the easy bit.

pfalcon, 2015-10-24:
As an update, I experimented with writing own minifier, and actually kinda have it, but now the biggest issue is to test that it produces correct code. And given #1527, maybe it's no longer that important.

pfalcon, 2016-02-01:
Closing this as no longer relevant per above comment.

closed 2016-02-01 by pfalcon