micropython/micropython #1310: I2C crashes Pyboard

peterhinch, 2015-06-06:
The following code results in a crash. With the timeout argument to mem_read() removed it runs. It also runs with the disable and enable IRQ statements removed. The hardware used is an InvenSense MPU9250. Code run from rshell.
```python
import pyb
import micropython
micropython.alloc_emergency_exception_buf(100)

class foo(object):
    def __init__(self):
        self.interface = pyb.I2C(1, pyb.I2C.MASTER)
        self.buf1 = bytearray([0])

    def read(self, buf, memaddr, devaddr):
        irq_state = pyb.disable_irq()
        self.interface.mem_read(buf, devaddr, memaddr, 5000)
        pyb.enable_irq(irq_state)

    def getacc(self, timer):
        self.read(self.buf1, 0x75, 105)

myfoo = foo()

tim = pyb.Timer(4, freq =10)
tim.callback(myfoo.getacc)

print("Waiting")
pyb.delay(3000)
tim.callback(None)
print(myfoo.buf1[0]) # Expected result 113
```

dpgeorge, 2015-06-06:
Where does it crash?

The timeout argument to mem_read is keyword-only, so it's not expected to work.  I would say that call to mem_read is raising an exception during the interrupt, with irqs disabled, and that leads to a bad state.

peterhinch, 2015-06-07:
It also crashes if I issue the timeout as a keyword argument
```python
self.interface.mem_read(buf, devaddr, memaddr, timeout=5000)
```
As for where the crash occurs, I'm not sure how I could establish this apart from by commenting out statements, an approach which points to mem_read() being the culprit.

dpgeorge, 2015-06-07:
First of all, your memaddr and devaddr args for read() are the wrong way around, but I assume that's also a typo.

Then there are 2 reasons for the lock-up:

1. Heap memory is being allocated in the foo.read() function, I think because this function has a Python stack size which is over the limit for putting it on the C stack.  So it tries to allocate heap for it and fails.  If you put read()'s code directly in getacc() then it works.

2. An exception (MemoryError) is being thrown with irqs disabled (by your pyb.disable_irq()) *and* from within the Timer callback.  This means that irqs are never re-enabled and the USB no longer functions.  If you call foo.getacc() directly from the REPL then irqs will be automatically re-enabled for you.  But since the call comes from a callback the irqs remain disabled.

Point 1 can be improved by increasing the limit on the amount of Python stack that can be allocated on the C stack.  We should probably do this because your code is pretty simple and it would be nice if it didn't need heap.

Point 2 can be improved if we re-enable IRQs if a callback raises an exception.  But I don't think that's a robust solution, and may lead to other unforeseen problems.

The main question is: why do you need to disable irqs during the callback?  Would raising the interrupt priority of the callback be adequate?

dpgeorge, 2015-06-07:
A related issue is that USB CDC read/write assumes irqs are enabled, so printing to the USB CDC could in some cases hang if irqs are disabled.  In 0aa5e7500047d869cc78e8914fc08b5d6a3f5994 I patched the USB CDC driver so that it doesn't make this assumption and can now work ok if irqs are disabled.

fojie, 2015-06-08:
Hi, I think this might be related to my problem with a small timer callback that calls two functions (one *scan_keys* [in-line assembler, should not be the culprit], and one method *fifo.put*).  See <https://github.com/fojie/micropython--In-Out/tree/master/Keypad>
The callback itself only uses **globals**, fifo.put uses 3 integers (on stack?) and sets the IRQ mask for very short time. This can run for hours, but **occasionally when key pressed** I get: 

<pre><code class="python">uncaught exception in Timer(4) interrupt handler
		NameError: name 'scan_keys' is not defined
		Traceback (most recent call last):
		  File "test_keypad.py", line 48, in <module>

def scan_timer_callback(timer):  # Called @50–100 Hz by Timer
	"""Callback for Timer. Scan keypad and debounce. Takes 13 / 26 µs [no key] / [key pressed]."""

	global key_last, key_old, key_flag, key_buf

	key_last = scan_keys()      # Call Assembler routine, returns key-# (0 if no key)
	if key_last == key_old:
		if key_flag == 2:       # EXPORT key
			fifo.put(key_buf, key_last)
			key_flag = NEW      # = 3 One loop more so we don't come back here (When Consumer clears key_flag)
		elif key_flag < 2:
			key_flag += 1       # One loop more
		else:
			pass                # Do nothing: Wait until keys change
	else:
		key_flag = NONE         # = 0 New key or key released
		key_old = key_last      # Key changed!: wait till next scans and check if stable

class FIFO() (excerpt)
def put(self, arr, data):
		""" Put new data into buffer and increase nr_bytes."""
		arr[0] = data                   # Always update peek = last input data
		s = arr[3]                      # nr_data = already in arr
		if s < arr[4] - 5:              # Check if room for more
			p = arr[1]                  # Yes, so Fetch put_ptr
			arr[p] = data
			p += 1
			if p >= arr[4]:
				p = 5                   # Make circular (skipping pointers in beginning)
			arr[1] = p                  # Update put_ptr -> next position to put data
			''' Protect the critical section: update nr_data (do not use 's' since it could be old by now)'''
			int_stat = disable_irq()    # We don't want an interrupt just now
			arr[3] += 1                 # Publish new data
			enable_irq(int_stat)        # Restore previous mask
		else:
			pass                        # We are full, do nothing</code></pre>

peterhinch, 2015-06-08:
@dpgeorge Sorry, my variable naming is confusing but the correct arguments are being passed to mem_read() and the code returns the expected value from the hardware.

The reason for disabling interrupts is as follows. I'm writing a driver for the MPU9250 IMU based on that for the MPU9150. We experienced unexplained issues with the latter driver. In an environment where pin interrupts occurred with a fairly high frequency we encountered occasional permanent I2C bus lockups. We discovered that the problem could be fixed by disabling interrupts for the duration of I2C read and write operations. Accordingly we provided the driver object with a disable_interrupts property to provide this option to the user.

While extensive testing showed it was reliable, the cause was never explained and I fully accept it's a kludge. If the underlying cause were to be fixed I'd rather not meddle with IRQ status.

As a general comment, writing code capable of being used in an interrupt handler seems challenging: to the naive user it's far from clear when the heap is liable to be invoked. I don't know if the answer is improved documentation, but I have found it somewhat challenging :)

dpgeorge, 2015-06-08:
> We discovered that the problem could be fixed by disabling interrupts for the duration of I2C read and write operations.

This sounds very similar to issue #685 where SPI was getting interrupted.  The fix there was to use DMA for SPI transfers (when irqs are enabled).  I would like to do the same for I2C, ie use DMA when possible.

I also just realised that with irqs disabled the I2C timeout does not work (because systick never increments with irqs disabled).

> If the underlying cause were to be fixed I'd rather not meddle with IRQ status.

The I2C read/write code at the moment just uses simple polling and it's possible that you miss a byte if you get interrupted for too long during an I2C transfer.  So using DMA should fix this in a more comprehensive way.

> As a general comment, writing code capable of being used in an interrupt handler seems challenging: to the naive user it's far from clear when the heap is liable to be invoked.

Yes I agree.

> I don't know if the answer is improved documentation, but I have found it somewhat challenging :)

Here are some things that we can try:

1. In places where there is a choice to allocate on the C stack or heap, make it always allocate on the C stack if the heap is locked.  This should be pretty robust since there is decent stack overflow checking, and in principle the only limit to allocating on the C stack is how much total stack space you have (so any restriction that we have currently in the code is rather arbitrary).  I think doing this will eliminate a lot of the cases that are hard to understand why heap is being allocated.

2. Provide a new function decorator "@micropython.noheap" which does one or more of the following:
- (documentation) serves as documentation that this function (and all its called children) do not allocate any heap memory
- (static check) allows the compiler to raise an error if the decorated function emits any bytecodes that it knows allocate heap memory
- (runtime check) allows the runtime to lock the heap on entry to the function to guarantee that it doesn't allocate, allowing you to easily check that the function works by executing it in some test cases

3. Provide functions to lock/unlock the heap so you can guard parts of your code from allocating on the heap (and also run tests to show that no heap is being allocated).

peterhinch, 2015-06-08:
> I would like to do the same for I2C, ie use DMA when possible.

That sounds ideal. At the time we tried to find a way to reliably reproduce the bus lock issue so we could raise it on Github. Alas we failed to find a way to replicate the bug short of an unreasonable amount of code and hardware. But running I2C by polling with interrupts running strikes me as hairy and hopefully DMA will fix it.

The ideas about the heap sound promising. I have been pushing the bounds of MicroPython interrupt handlers to establish what is and is not feasible within them: an educational experience :)

danicampora, 2015-06-08:
> In places where there is a choice to allocate on the C stack or heap, make it always allocate on the C stack if the heap is locked. This should be pretty robust since there is decent stack overflow [...]

+1 :-)

> Provide a new function decorator "@micropython.noheap"

+1 Sounds like a great idea as well.

dpgeorge, 2015-06-08:
> In places where there is a choice to allocate on the C stack or heap, make it always allocate on the C stack if the heap is locked.

There is only 1 place in the code where this occurs and I've now changed it so that it falls back to C-stack allocation (ie alloca) if heap allocation fails; see 6e56bb623c66007cd3fe6e0a48c3af9f1e0814fc.  This should mean that code that doesn't look like it allocates heap, probably doesn't.  In particular the code that is the subject of this issue should now run correctly.

peterhinch, 2015-06-09:
I can confirm that the code now runs correctly.

Should the bug discussed above (whereby I2C occasionally fails in the presence of hardware interrupts) be raised as a new issue? I must warn it is very hard to replicate.

dpgeorge, 2015-06-09:
> Should the bug discussed above be raised as a new issue?

No, it's the root cause of this issue so can stay here, and this issue stay open.  I'm working on I2C DMA, but it might take some time :)

dpgeorge, 2015-06-10:
DMA for I2C now implemented; see 76285469d3e64ffe0312f58fa1f6270f715c842c.

@peterhinch are you able to test this new code to see if the I2C runs reliably without disabling irqs?

peterhinch, 2015-06-10:
Great news! I should be able to test it tomorrow (Thursday).

fojie, 2015-06-11:
I have tested "pybv10-2015-06-11-v1.4.3-147-gcfcf47c.dfu" and there seems to be a problem with i2c bus nr 2 (it seems dead):
<pre><code class="python">i2c = I2C(2, I2C.MASTER)
i2c.scan() returns []

while with the "v1.4.3 on 2015-05-17" firmware I get:
i2c.scan() returns [24, 27, 32, 33, 64, 92, 104, 119]</code></pre>
bus nr 1 seems working, but it's always easier to proof that something is NOT working, than working…

peterhinch, 2015-06-11:
@fojie I can't replicate your problem here. With the same build of firmware I get
```
>>> import pyb
>>> a = pyb.I2C(2,pyb.I2C.MASTER)
>>> a.scan()
[12, 104]
```
@dpgeorge I'm currently failing to replicate the fault with firmware I built from source on 9th June. It always was a hard one to provoke. The new build does work, but in the absence of a comparison with a known faulty build it isn't the most convincing demonstration. I'll do some more testing.

peterhinch, 2015-06-11:
@dpgeorge The only way I could replicate the fault was by reverting to a firmware build from the time when I was working with @turbinenreiter to find a workround. With that build I had an I2C failure within one second. Both the 9th June and 11th June builds are reliable in my testing. So I can only assume that the underlying cause of the problem we had encountered was addressed between 26th Sept 2014 and 9th June.

In any event I've run the new build with two different I2C devices. I've run both I2C interfaces, and run for some time in the presence of four continuously occurring pin interrupts and it has been entirely reliable. As far as I'm concerned it's fixed :)

dpgeorge, 2015-06-11:
It may be that both I2C busses running at the same time causes issues... they both use DMA 1 Stream 7 for transfers so can't run concurrently (which can only happen if one is doing a send and gets interrupted by an IRQ which then does a send on the other bus).  But the scan() method should still work because it doesn't use DMA.

@peterhinch thanks for the extensive testing, it's good that both busses run reliably.  I think then this issue can be closed.  @fojie: please open another issue specific to your problem if you can find a (semi) reproducible bug.

closed 2015-06-11 by dpgeorge

fojie, 2015-06-11:
OK, I will have to more testing. Hopefully there's some simple explanations (like error from my side). 

@dpgeorge Since I think I have the concurrent situation you describe; Do you mean that concurrent sends should be avoided for ch1/ch2, but it's ok with ch1/ch1 and ch2/ch2 (ch2 gets interrupted for another ch2 send)? Or are all concurrent sends to be avoided?

dpgeorge, 2015-06-11:
@fojie wow, you are playing with fire doing concurrent sends!  There is no way it'll work with ch1/ch1 or ch2/ch2 because the bus is a physical thing that can't be shared: if you're in the middle of an I2C transaction and get interrupted and then try to perform another transaction on the same bus then that won't work no matter how hard you try.  Working on ch1 then getting interrupted and working on ch2 can in principle work (you'll definitely need to use DMA here) but not in practice at the moment because the I2C busses share the same DMA stream.  They share because one of the other available streams is used for the DAC... the DAC is something that needs a dedicated stream because it's most likely going to be running in the background, whereas I2C transfers generally finish quickly.

fojie, 2015-06-12:
Thank you for making it clear, so even I understand! I was a bit optimistic. I'll try to find another strategy.