micropython/micropython #1311: ESP8266 WiFi esp.connect() causes crash with latest builds

owens-bill, 2015-06-07:
Building the current firmware with current esp-open-sdk succeeds, but the resulting code is crashing on the esp.connect() call. For example:

Micro Python v1.4.3-128-gb4a41a8 on 2015-06-06; ESP module with ESP8266
Type "help()" for more information.
>>> import esp
>>> esp.phy_mode()
1
>>> esp.scan(print)
>>> (b'linnjk', b'df\xb3\xbc>\xb0', 6, -88, 4, 0)

>>> esp.connect('foo', 'bar')

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32280, room 16 
tail 8
chksum 0x89
load 0x3ffe8000, len 1380, room 0 
tail 4
chksum 0x8f
load 0x3ffe8570, len 20672, room 4 
tail 12
chksum 0x7d
csum 0x7d
??ՁMEM CHECK FAIL!!!
?rl??rl??b
>>> 
Micro Python v1.4.3-128-gb4a41a8 on 2015-06-06; ESP module with ESP8266

This is an ESP-07 module on a Baoshi breadboard adapter, with an external power supply to avoid any issues with voltage sag (5 volt/3 amp supply feeding the regulator on the adapter). I get the same result with an ESP-01 on a breadboard adapter powered by a PL2303 serial cable. The crash also occurs when using legit SSID and password, on an 802.11g network with WEP2 that I could previously connect to.

chinadsf, 2015-06-09:
yes,same problem with ESP-01 module.

pfalcon, 2015-06-12:
Fixed in 431603ad696af720917bce7f5c2f908aad72780c. The regression comes from an untested patched merged in 286ced4c2f800d0837397dd2a0c8deea8c7ade4e. @atalax : FYI.

pfalcon, 2015-06-12:
(And yep, I tested that .connect() actually works with latest esp-open-sdk).

closed 2015-06-12 by pfalcon

dpgeorge, 2015-06-12:
@pfalcon I don't think you pushed the patch.

danicampora, 2015-06-12:
> @pfalcon I don't think you pushed the patch.

I was about to say the same thing...

pfalcon, 2015-06-12:
Now done.