micropython/micropython #1313: esp.scan() changes WiFi operational mode

owens-bill, 2015-06-10:
While reading modesp.c I discovered that esp.scan() calls wifi_set_opmode to change the operational mode of the radio to 'station mode', disabling the Soft AP function:

```
STATIC mp_obj_t esp_scan(mp_obj_t cb_in) {
    MP_STATE_PORT(scan_cb_obj) = cb_in;
    wifi_set_opmode(STATION_MODE);
    wifi_station_scan(NULL, (scan_done_cb_t)esp_scan_cb);
    return mp_const_none;
}
```

That appears to be taken from an example in the ESP SDK docs. However, this has two side effects; it changes the operational mode and makes 'station mode' the new bootup default. It would be more graceful to have the function get the current mode (with wifi_get_opmode), then change just the current setting (wifi_set_opmode_current), and change it back afterwards. It seems like this version does the trick, but I don't feel confident enough in my C skills to be sure (and definitely not in my git skills to submit a pull request!)

STATIC mp_obj_t esp_scan(mp_obj_t cb_in) {
    mp_uint_t current_mode;

    MP_STATE_PORT(scan_cb_obj) = cb_in;
    current_mode = wifi_get_opmode();
    wifi_set_opmode_current(0x1);
    wifi_station_scan(NULL, (scan_done_cb_t)esp_scan_cb);
    wifi_set_opmode_current(current_mode);
    return mp_const_none;
}

I made the switch from STATION_MODE to the hex literal because I couldn't figure out where STATION_MODE is defined; it seems to be in the SDK somewhere rather than in the MicroPython code, so I wasn't sure whether it was robust against changes to the SDK.

owens-bill, 2015-06-10:
Having read the developer workflow and learned something about rebase, I decided to try a pull request after all ;)

owens-bill, 2015-06-12:
@pfalcon requested a test of esp.scan() in different modes (without changing to STATION_MODE before the call to wifi_station_scan). The answer is that the scan appears to work normally in station plus AP mode (what the SDK calls STATIONAP_MODE, 0x3) but the scan doesn't return anything in AP-only mode (SOFTAP_MODE, 0x2). I could not see any difference between STATION_MODE and STATIONAP_MODE, though the scan typically doesn't return all of the WiFi networks that are available so the results do vary from call to call.

owens-bill, 2015-06-12:
One other thing, just to be clear - I wasn't suggesting in my original that STATION_MODE doesn't need to be set; I think it is clear that it does, based on the example code in the SDK docs:
```
void to_scan(void) { wifi_station_scan(null,scan_done); }   void user_init(void) {  
wifi_set_opmode(STATION_MODE);  
system_init_done_cb(to_scan);   }
```
I just didn't want the opmode change to be invisible and persistent, in case someone is trying to use one of the other two modes and isn't aware of this side effect of the scan method.

pfalcon, 2015-06-13:
@owens-bill : Thanks for your contributions! And I see that they improved since this initial ticket, but I'd still like to go over some points from this initial one. I hope you will find that as desire to provide explicit feedback on "do's" and "don'ts" rather than neat-picking.

> It seems like this version does the trick, but I don't feel confident enough in my C skills to be sure

Taking into account that your C is pretty OK, I read an implication that you didn't test that change - because after all, it either works as expected, or not. Proposing untested changes isn't a good pattern, and you actually saw a good display of that - .connect() crash affecting you actually came from another guy's untested changes. So, please take care to test things you propose (asking question before trying to implement it is ok, and for more or less substantial changes is encouraged, but then oftentimes nobody can give you full answer).

pfalcon, 2015-06-13:
> I made the switch from STATION_MODE to the hex literal because I couldn't figure out where STATION_MODE is defined; it seems to be in the SDK somewhere rather than in the MicroPython code, so I wasn't sure whether it was robust against changes to the SDK.

That's another anti-pattern to avoid - breaking one thing while fixing another. It's basic rule of thumb in programming - to avoid magic numeric values. And if you don't know how something works, one good approach is not changing it ;-). Even better approach is challenging oneself into getting to know how it works - just like you did, and +1 on that.

pfalcon, 2015-06-13:
Now about the code specifically. First thing to note is that "scan" operation is asynchronous - you request it, and continue with the rest of your code, while results are delivered elsewhere. So, let's look:

~~~~
    wifi_set_opmode_current(0x1);
    wifi_station_scan(NULL, (scan_done_cb_t)esp_scan_cb);
    wifi_set_opmode_current(current_mode);
~~~~

So, you set suitable mode, request operation, and then immediately reset mode before operation is done. That's not supposed to work as you expect - either scan won't work well, or mode won't be reset. Or if you test it really well, it may work for you, but not for somebody else. Or it may break with SDK upgrade. So, I'm afraid, applying this patch won't do any good long-term.

pfalcon, 2015-06-13:
From https://github.com/micropython/micropython/pull/1315#issuecomment-111688840 :

> danicampora commented 3 hours ago
>I think the .scan()function should NOT change to Station mode under the hood. When the WiFi radio is configured in AP mode, it cannot scan for networks, that's correct behaviour, and so, .scan() should either return an empty list (as it does now), or throw an exception, perhaps the latter is better because it makes clear that something is not right. It's the responsibility of the user to select the correct mode before scanning for networks.

I think what @danicampora says makes good sense (thanks for keeping an eye on other ports too, Daniel!). And with your testing, @owens-bill, (thanks for it!) we know that AP mode indeed doesn'[tw ok for esp8266, so I'd re-do patch as @danicampora - throwing an exception if current mode is AP (returning empty list would rather confusing - run on a fresh module, I'd think it's a broken one for example).

owens-bill, 2015-06-13:
@pfalcon Thanks for the detailed feedback - that's exactly what I need to improve my understanding. I'm happy to learn from it for as long as your patience lasts ;) To hit just a couple of things, I did test the code, on several different ESP modules (01, 07 and 12 to be exact; Olimex and SparkFun just arrived yesterday). The misunderstanding about 0x01 vs. STATION_MODE was because I hadn't yet discovered the process of taking SDK constants and renaming them for MicroPython; I still don't know whether there's a pattern to doing that, though I took my best guess with the operational modes for PR#1318. I've learned to read the SDK more carefully (docs and code) and compare better with what's already in MP.

As for the core of the problem, thanks for pointing out that this change won't do any good ;) Definitely need to discard this PR and rewrite. I believe I understand how to generate the exception; something like:
```
nlr_raise(mp_obj_new_exception_msg(&mp_type_OSError, "Scan not supported in AP mode"));
```

And finally, I believe that most if not all of the ESP modules come 'out of the box' in STATIONAP_MODE, which should allow esp.scan() to function normally even if the user doesn't change the mode (once they have a Pythonic way to change the mode ;)

pfalcon, 2015-06-13:
~~~~
nlr_raise(mp_obj_new_exception_msg(&mp_type_OSError, "Scan not supported in AP mode"));
~~~~

That's correct!

owens-bill, 2015-06-13:
Ok, one more question while I test the new version: should I delete PR#1315 and start over, or change the commit to the new version once it's ready?

pfalcon, 2015-06-13:
Whatever is easier for you, just add a comment if you push to an existing PR - github doesn't send notifications of those.

owens-bill, 2015-06-14:
After yet another false step (not understanding how to squash commits - now fixed) I believe that PR #1315 is ready to fix this issue.

owens-bill, 2015-06-14:
Resolved by PR #1315.

closed 2015-06-14 by owens-bill