micropython/micropython #1327: stmhal: rtc.datetime() throws an exception if called without arguments

owens-bill, 2015-06-14:
The [documentation](http://docs.micropython.org/en/latest/library/pyb.RTC.html) for pyb.RTC on the pyboard says that without any arguments the method will return the current time as an 8-tuple; with one argument (an 8-tuple) the RTC will be set. The comments in the code say the same:
```
/// With no arguments, this method returns an 8-tuple with the current
/// date and time.  With 1 argument (being an 8-tuple) it sets the date
/// and time.
```
However, the code disagrees:
```
mp_obj_t pyb_rtc_datetime(mp_uint_t n_args, const mp_obj_t *args) {
    if (n_args == 1) {
        // get date and time
        // note: need to call get time then get date to correctly access the registers
```
and:
```
MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(pyb_rtc_datetime_obj, 1, 2, pyb_rtc_datetime);
```
The behavior is consistent with the code (naturally):
```
Micro Python v1.4.3-159-g3ce212e on 2015-06-14; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import pyb
>>> rtc = pyb.RTC
>>> dir(rtc)
['info', 'datetime', 'wakeup', 'calibration']
>>> rtc.datetime()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing 1 required positional arguments
>>> rtc.datetime('foo')
(2014, 1, 1, 3, 0, 37, 49, 203)
>>> rtc.datetime((2015, 6, 14, 7, 0, 0, 0, 0))
(2014, 1, 1, 3, 0, 39, 11, 86)
>>> rtc.datetime('foo',(2015, 6, 14, 7, 0, 0, 0, 0))
>>> rtc.datetime('foo')
(2015, 6, 14, 7, 0, 0, 5, 101)
>>> rtc.datetime()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing 1 required positional arguments
```
Am I reading this wrong? I made a simple change (three lines) to rtc.c and produced the behavior as documented, but I don't understand why it would have been different in the first place...

dpgeorge, 2015-06-14:
It's correct as is.  You need to instantiate an instance of the RTC class to use it:
```python
rtc = pyb.RTC() # note parenthesis!!
rtc.datetime()
```
The first (hidden) argument is the instance.  The second optional argument is the 8-tuple to set the datetime.  See https://github.com/micropython/micropython/blob/master/tests/pyb/rtc.py for example usage.

(It just so happens that the RTC methods don't need any info from the instance so passing 'foo' as the instance is good enough to make it work.)

So the real question is: why did you create an RTC instance using `pyb.RTC` instead of `pyb.RTC()`?  Was the documentation not clear in this respect?

owens-bill, 2015-06-15:
I created the instance that way because that's how the ESP8266 port works:
```
Micro Python v1.4.3-155-g5047b9d-dirty on 2015-06-13; ESP module with ESP8266
Type "help()" for more information.
>>> import pyb
>>> rtc = pyb.RTC()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object not callable
>>> rtc = pyb.RTC
>>> dir(rtc)
['datetime', 'memory']
>>> rtc.datetime()
(2014, 1, 1, 2, 4, 53, 49, 398)
>>> rtc.datetime('foo')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: expected tuple/list
```
But of course you're right, the docs for the pyboard correctly show "rtc = pyb.RTC()". Of course, it does seem as though the two should agree; there's another difference I've run into and I was going to open an issue on that, but got distracted by this one.

dpgeorge, 2015-06-15:
Yes, the esp port needs to be changed to have compatible behaviour.

owens-bill, 2015-06-15:
Got it. I will close this issue and recast it as changes needed to the esp8266 version, and include the other discrepancy I've found.

closed 2015-06-15 by owens-bill